bambulab/BambuStudio #3111: Custom Filament Does not Show in Device Tab and Printer LCD Menu

clstr, 2023-12-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 Pro

### Printer

X1C w/ AMS Firmware: 1.7.0 

### How to reproduce

1. Create custom filament per wiki (https://wiki.bambulab.com/en/bambu-studio/create-filament)
2. Select Filament in Device Tab

### Actual results

After selecting custom in device tab it disappears and shows a "?".

[image]

I can still run calibration, but i have to select the custom filament because the default is ?

[image]

[image]

It does save the K Factor results to the slicer, but does not get properly shown in the Device Tab. 

Additionally, the drop down menu on the X1C does not show any custom filament. I made about three custom filament.

[image]

[image]

### Expected results

Custom filament should be populated correctly in the Device Tab of the slicer and should show up on the printer drop down menu.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13677222/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DanBao-Bambu, 2023-12-15:
This issue requires the printer's log. Can you export the printer's log and send it to me?

QingZhangBambu, 2023-12-15:
#3807

clstr, 2023-12-15:
@DanBao-Bambu Log from printer is too large for github, it is 416MB, limit on github is 25mb.

I don't know where I can upload. 

Beside this i was able to get the custom filament to show up but I had to turn on Cloud mode, I was using LAN Mode. 

1. I tested again by placing printer in LAN mode again create a new custom filament in Bambu Studio. 
2. On the Device Tab it still shows up as "?" and the printer LCD the new custom filament does not show up. 
3. I placed the printer back on cloud mode and the custom filament now shows up. 

This is seriously flawed behavior and I should not need to place printer on cloud mode to use custom filament.

pelrun, 2023-12-15:
As a new feature it's not entirely unsurprising that they integrated it with the Cloud first, since that's their primary recommended workflow. To BL's credit they've been actively adding all the major features to LAN Only mode, and I can only hope the plan is to continue that with the custom presets.

It'd be nice if the documentation made it clear that it's Cloud only for the moment, just to eliminate confusion on our end.

DanBao-Bambu, 2023-12-15:
@clstr Yes, the printer cannot receive the newly created custom Filament in LAN mode. The current solution is to enable cloud mode connection to the printer, enter the Filament selection interface of the printer, and after receiving the custom Filament information, the printer can switch back to lan mode.
I know it's tedious, I'm sorry.

tokamac, 2023-12-15:
[DanBao-Bambu](https://github.com/DanBao-Bambu) the issue occurs in X1C printers that are NOT in LAN mode.
By the way you keep continue answering here, but this thread is a duplicate of prior #3087 as already pointed out by @QingZhangBambu above.

scottbmc, 2023-12-15:
I can confirm that i have the same problem with my X1C, fully on cloud mode. I Can create custom Filaments but as soon as i select them in the "device" tab, they show a questionmark if i select a different slot or try to set the same filament brand with a different color for a different slot.

OzDruiD, 2023-12-15:
This is very strange, as I cannot replicate the fault, in full cloud only mode.
I have now created multiple new custom filaments., as primary and apepar on device tab and printer
Even creating a preset and then calibrating that preset (instead of the master) and then those also appear on the printer (k prorfile etc)

tokamac, 2023-12-15:
Just a thought: Could OrcaSlicer, which allows for storing (and accessing for modifications) the K value directly in the "Basic information" of filament settings in the slicer, have messed with the printer code? I sometimes use it besides BambuStudio.

DanBao-Bambu, 2023-12-16:
@tokamac @scottbmc If there is a "?" issue with the cloud connected printer, the printer's log is indeed required. I know it is quite large, so you can use the "Google Drive" cloud service to upload the log and share it with me for connection. This is very important to me, thank you.

@tokamac Thank you for providing the reference. I will try and confirm if this is relevant later. Please confirm whether the preset calibration you mentioned is pressure calibration or flow calibration.

scottbmc, 2023-12-16:
Here is my printer logfile. Hope this helps.

https://drive.google.com/file/d/1Xcht78mj5yOm0ItprGo4PixJr-UnER-M/view?usp=sharing

DanBao-Bambu, 2023-12-18:
Please try entering the material tab of the printer screen after creating the Filament and entering a slot to set material information. After returning to the Device interface of Studio, setting Filament should not cause the above problem. If there are any issues, please continue to contact me and I will continue to track the issue until it is resolved.

tokamac, 2023-12-22:
Latest X1C FW **01.07.01.00** does NOT correct the issue :(
BTW where are the [release notes](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)? It's been hours the firmware has been out and they are not available! Was this release (that occurs only 11 days after the previous FW update) a hot fix or a mistake?

broncosis, 2024-01-05:
I can confirm  the same issue in the Linux release of Bambu studio and my machine is fully connected to the cloud 

linux Mint with Bambu lab 1.8.2.56
x1c  01.07.01.00 
ams 00.00.06.40

DawtCom, 2024-01-08:
Just adding that I am having the same issue just to raise the issue's importance.

antgir1, 2024-01-09:
Hi, same problem here, Bambu Studio version 1.8.2.56. 

The filament selection of custom filament  in device always disappear !

I'm also confured to use the CLOUD.

Here's a little video of the problem: https://share.icloud.com/photos/0a6Fx2K84KpnZwf7VOCLzGzIw

Hope this help

dperlman, 2024-03-23:
I am having the same issue and my device is on the default cloud setting. The behavior I see is exactly the same as the screen recording shown in this forum thread: https://forum.bambulab.com/t/newest-bambu-studio-setting-new-brand-filament-will-delete-itself/42680 

The workaround of adding through device screen does work as a workaround but there is still a bug happening here. Happy to share any logs as needed to help troubleshoot this because it is VERY annoying!

DanBao-Bambu, 2024-03-24:
As far as I know, there is currently no printer version available for A-series machines that supports custom Filament.
If the printer is an X1C series, you can try to solve it by following the steps below.
1. Ensure that the user's preset cloud synchronization is enabled
2. After creating a custom Filament, go to the printer screen and enter the Filament settings interface to check if your custom Filament already exists. This ensures that the printer has been synchronized from the cloud to the newly created custom Filament.
3. Custom Filament can be set up normally on Studio.
I know that setting up the printer is quite complicated, Studio is about to release a new beta, which does not require operation on the X1C screen, but requires a new software version of X1C to work with.

For specific details, please refer to the following link:
https://wiki.bambulab.com/en/software/bambu-studio/custom-filament-issue

If you are still unable to set custom Filament, please send Studio and printer logs. Thank you very much.

kmeldner, 2024-04-01:
I have the same problem.
I have followed instructions from 
https://wiki.bambulab.com/en/software/bambu-studio/custom-filament-issue
and no effect.

Can someone explain in simple understandable step by step instruction, like for dummies, how do I create custom filament and set it so it is available to be assigned to AMS slot in the printer?

DanBao-Bambu, 2024-04-15:
@kmeldner 
- Confirm that the printer model is X1C and update the printer version to the latest version (as far as I know, custom materials have been supported since 1.7.2, but there are some issues, it is best to use the latest version).
- Use the latest Studio version, release version, or Beta version (starting from version 1.8, custom Filament is supported, and Beta version provides more convenient settings). It is recommended to use Beta version.
- Next is the operation on Studio: Open the Preferences interface and enable the "Auto sync user presets" option.
[image]
- Follow the steps to create the custom Filament you want.
[image]
[image]

- After creating your custom Filament, switch to the Device interface and connect to the X1C model's machine (currently P series and A series do not support custom Filament).
- The next step is to operate the printer and enter the Filament settings interface.
[image]
- Then the printer will enter the specific settings interface. After entering this interface, the printer will obtain custom materials from the cloud. You can check whether the custom Filament you created is in the list by viewing the dropdown menu. If it is, you can select it or select the Filament you created in Studio. Setting the Filament in the future will not be used for printer confirmation because the printer has already obtained information about the Filament. If the custom Filament you created is not in the drop-down list, you can choose to try logging in to your account again on the printer, and the printer will also receive your Filament list once.
[image]
- At this point, the custom Filament can be successfully set on Studio, but when you create a new custom Filament, you need to go to the printer again to confirm.
However, the compatibility between Studio 1.9 Beta and future printer versions can achieve successful material settings on Studio without the need for confirmation at the printer end.