bambulab/BambuStudio #3112: Unable to reduce the Extruder Temp after

Buffalo1John, 2023-12-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro 2h2

### Additional system information

Processor	Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
Installed RAM	32.0 GB (31.6 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

P1S

### How to reproduce

This bug requires that the filament fail to load because any number of issues - broken filament in AMS, clogged extruder, etc.

Attempt to load filament, extruder temp goes to 250C (this is problematic to anyone who uses just PLA like me, but that is not the bug).

Load fails and message is generated both on Banbu Studio and on from panel. Problem is understood and the remedy requires some time to implement, so try to change the extruder temp (no need to keep it powered at 250C). On front panel, unable to make any change in temp. You can change the target, but pressing OK or return does nothing - still at 250C. Try to do it in Bambu Studio under the device tab, you see 250C / 250C and if you click on the second 250C, you can enter a value. Enter 0 and hit return and it pauses and goes back to 250C. Do that a few times, hit retry or confirm (either one) on message or leave the message as is, entering any value does not work. After a few minutes, it magically changes the value to 42C as the target and the extruder finally cools to 42C. You can try to change 42C to 0C and no combination of anything clicked or not makes 0C stick. Removing power to the system will clear the problem, but that also drops power to the fan that keeps the extruder from getting clogged (Heat Creep is written about in the WIKI), so that is a horrible solution. I now have to take the extruder apart to fix this very problem.

There should be a way to drop the temp under software control, either the front panel or Bambu Studio. All my other printers have this ability, and this is extremely useful and keeps the components from getting further damaged. It would also be enice if we could have control over the 250C, that is excessive. Yes, I know you could have had ABS loaded and purging it so you can do PLA requires 250C, but if I had only PLA and I'm changing filament colors, I don't need a 250C purge - especially if any goes wrong loading (I had trouble getting the feed point pushed back on the ams because it is such a short throw and is very stiff, so I bent the end of the new filament and it got stuck and broke off - just one example of many)

### Actual results

Can't cahnge the 250C extruder target and it is stuck that way.

This happens on any project and on no project - just setting the filament in the device tab.

### Expected results

You should be able to lower the extruder temp.

### Project file & Debug log uploads

This problem is not generating a log nor is it dependant on a project.
[BB_CommonAXA-Body.zip](https://github.com/bambulab/BambuStudio/files/13678466/BB_CommonAXA-Body.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

walterwongbbl, 2023-12-18:
@sodachen22 please check

Sonicboom247, 2023-12-31:
I sort of agree here, one option would be to add a filament change temp based on the type of filament detected or if there is a clog, then go up to 250c. Otherwise, those users that print with PLA would see something around the 215-225 degree range. With a desirable temp per filament, like TPU, Wood, etc.