bambulab/BambuStudio #3113: Check marks for flush options missing

Elan114, 2023-12-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

1)Right click on object
2)Go to Flush Options
3)Select any of the 3 options (no check mark appears to show that the selection is active)

Repeat steps and the selection indicator is still not there. Checking the preview you can see whether it is flushing to the object, the supports or infill.

### Actual results

You can't tell if a flush option is activated by looking at the right click menu

### Expected results

There should be a check mark or some indicator showing that the selected flush option is active

### Project file & Debug log uploads

[Patriots.zip](https://github.com/bambulab/BambuStudio/files/13678668/Patriots.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-12-15:
if you don't mind, could you post a pic of the that?

Elan114, 2023-12-15:
[image]
[image]
There is no indicator next to the flush values to show what is active and what is not. I know I used the feature last week in an older version of Bambu Studio and you could tell what was and what was not activated

QingZhangBambu, 2023-12-15:
Ok，i get it. it should be a new feature. i will add it to our feature pool

QingZhangBambu, 2023-12-15:
> Ok，i get it. it should be a new feature. i will add it to our feature pool

i mistake, it should be a bug

LiZ-Li-BBL, 2024-01-05:
Hello,
This issue has been fixed. Please wait for the next release soon.

closed 2024-01-16 by LiZ-Li-BBL