bambulab/BambuStudio #3114: GCode thumbnail not displaying correctly

TaylorN15, 2023-12-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

1. Slice file
2. Send to printer

### Actual results

The preview thumbnail is shown with a checkered background, like a PNG file saved as a JPG.
[image]

### Expected results

Checked background should not be visible. This has only started in v 1.8

### Project file & Debug log uploads

[files.zip](https://github.com/bambulab/BambuStudio/files/13678706/files.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-12-18:
Hi. @TaylorN15 
This is a new feature we have implemented because in dark mode, the model and background color blend together, making it difficult to distinguish. Therefore, in dark mode, we have changed the background color to a grid background."

TaylorN15, 2023-12-19:
> Hi. @TaylorN15 This is a new feature we have implemented because in dark mode, the model and background color blend together, making it difficult to distinguish. Therefore, in dark mode, we have changed the background color to a grid background."

I didn't see anything about this in the release notes, so I assumed it was a bug with rendering the image. Doesn't look very nice, but if it's not a bug, then I will close this - thanks

closed 2023-12-19 by TaylorN15