bambulab/BambuStudio #3117: Issues with modifier ordering when duplicating assemblies

romanhuantedesign, 2023-12-15:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Duplicating an assembly that contains a modifier.

Please see this reddit post for a detailed video description of the problem. I tried uploading the video here but it's too large.

https://www.reddit.com/r/BambuLab/comments/18ijhr3/issues_with_modifier_ordering_when_duplicating/

### Actual results

1) The new duplicated assembly is identical to the original, except the modifier is at the top of the list of parts instead of the bottom, making the modifier inactive. 

2) The modifier cannot be dragged to another place in the list the way that parts normally can. The only way to bring it to the bottom of the list is to change it to a part and then change it back to a modifier. 

3) Furthermore, I cannot rearrange parts in an assembly that is a duplicate of another assembly. I feel like this could be really useful if it was a sign of the duplicate being a parametric reference of the original. That seems not to be the case though. If I edit the original the duplicate doesn't change.

### Expected results

1) The new assembly should maintain the same relative positioning of parts and modifiers as the original. 

2) Modifiers should be able to be rearranged the same way that parts can. 

3) Allow these parts to be rearranged unless intentionally designated as a parametric reference.

### Project file & Debug log uploads

[Strawberry Demo.zip](https://github.com/bambulab/BambuStudio/files/13679471/Strawberry.Demo.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-12-15:
if you don't mind, could upload the log of studio to help us to target the issue.

romanhuantedesign, 2023-12-15:
> if you don't mind, could upload the log of studio to help us to target the issue.

Thank you for the reply! Sure thing, here you go. (: Also, I have an update. I just installed the latest version of Bambu Studio and it looks like the problem has been partially solved. When I copy objects, the order of the modifier still changes within the assembly, but the modifier seems to be active now at least. I still have no way of being able to rearrange it though. Does the order of the modifier in the list of parts affect how it is applied? For instance, if the modifier is on the top of the list, does it not affect any of the objects below it?

[Strawberry.Demo.zip](https://github.com/bambulab/BambuStudio/files/13689543/Strawberry.Demo.zip)

IanVS, 2024-04-22:
I have this same issue.  I'd like to print multiple copies of something that contains modifiers, but because the order of the modifiers changes to be on top, their filament selection is not used as it is overridden by the stl at the bottom.  Is there any workaround anyone has found?