bambulab/BambuStudio #3120: video playback fails for p1s on linux.

JeremiahSecrist, 2023-12-15:
### Bambu Studio Version

1.8.1.57

### Where is the application from?

Built from source by yourself or others

### OS version

NixOS 24.05.20231211.a9bf124 (Uakari) x86_64 

### Additional system information

Host: Framework FRANMDCP07 
Kernel: 6.7.0-rc5 
Packages: 1602 (nix-system), 1912 (nix-user), 35 (flatpak) 
Shell: zsh 5.9 
Resolution: 2256x1504, 1920x1080 
DE: Plasma 5.27.10 (Wayland) 
WM: kwin 
CPU: AMD Ryzen 7 7840U w/ Radeon 780M Graphics (16) @ 5.132GHz 
GPU: AMD ATI Phoenix1 
Memory: 22239MiB / 31287MiB 

### Printer

P1S

### How to reproduce

0. install nixos version of bambu studio via nix: nix-shell -p bambu-studio (just for note, this also happens on the official appimage for me)
1. Add printer to bambu studio
2. Go too device tab and click on printer
3. Try to view webcam and watch it fail with various error codes [2] being one of them
4. try on x1c to compare and watch it work on linux
5.  try watching the p1s and watching it work

### Actual results

[image]

### Expected results

The expectation is for the video to play on the P1S.

### Project file & Debug log uploads

[Uplo
[tutk_log_avapi_180376.txt](https://github.com/bambulab/BambuStudio/files/13680632/tutk_log_avapi_180376.txt)
[tutk_log_iotc_180376.txt](https://github.com/bambulab/BambuStudio/files/13680633/tutk_log_iotc_180376.txt)
ading tutk_log_global_180376.txt…]()

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-12-15:
Is the liveview of X1C working properly?

Please set envriment 'GST_DEBUG=4' and run Studio from console.
Then more log messages relate to GST Video Player will be dumpped on console.
Send the logs to us.

JeremiahSecrist, 2023-12-15:
x1c works fine by contrast on this very build.
[log.txt](https://github.com/bambulab/BambuStudio/files/13681143/log.txt)

bambu123, 2023-12-15:
No decoder available for type 'image/jpeg'
Install gstreamer1.0-plugins-good may help.

JeremiahSecrist, 2023-12-15:
> No decoder available for type 'image/jpeg'
Install gstreamer1.0-plugins-good may help.

trying now. waiting for it to finish building.

JeremiahSecrist, 2023-12-15:
Thanks that fixed it! before you close this i need to fix this on nixos side of things.

closed 2024-01-04 by JeremiahSecrist