bambulab/BambuStudio #3121: Cannot replicate support behaviour in v1.8.2.56 that was able to in 1.7.7.89

Cenwulf, 2023-12-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

n/a

### Printer

Bambu Lab P1S

### How to reproduce

1. Download project file: [huego_lithophane_bracket.zip](https://github.com/bambulab/BambuStudio/files/13683150/huego_lithophane_bracket.zip)
2. Open project in Bambu Studio v1.7.7.89 and observe support interface layers.
3. Open project in Bambu Studio v1.8.2.56 and observe missing bottom support interface layers (see bellow)

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[huego_lithophane_bracket.zip](https://github.com/bambulab/BambuStudio/files/13683336/huego_lithophane_bracket.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-12-16:
It looks like the problem with snug support. Try normal support with grid style.

Cenwulf, 2023-12-16:
> It looks like the problem with snug support. Try normal support with grid style.

Thanks but I'm just sticking with the previous release for now as it doesn't have this problem. My main aim was to highlight this undesirable (to me at least) change in support behaviour that's been introduced in the latest release.

tangketan, 2024-03-08:
Indeed. We'll fix it soon.