bambulab/BambuStudio #3122: Connectivity problem after update to 1.8.2.56

x1c3di, 2023-12-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro

### Printer

X1c combo

### How to reproduce

I updated  Bambu studio to V 1.8.2.56 when suggested by the app itself

### Actual results

The slicer updated and ran as usual, but when i open the 'Device' tab the slicer starts constantly connecting and immediately disconnecting with the printer at random intervals, and doesn't stop as long as the 'device' tab is open. Every time it happens, it triggers the AMS to slightly rotate one of the spools back and forth.

When the little orange tab comes on to tell you it failed to connect, it shows error code -6010

It is still able to send print jobs to the printer, but during the print , if i'm on the device tab it still keeps switching between connected and disconnected

In lan only mode everything works normally

Bambu handy does not seem to be affected

### Expected results

a stable connection

### Project file & Debug log uploads

[tutk_log_avapi_5468.txt](https://github.com/bambulab/BambuStudio/files/13683542/tutk_log_avapi_5468.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mezanu, 2023-12-15:
Exactly the same problem, BambuX1C AMS, European Region, Win11, latest firmware, after updating at the request of the program, the connection is intermittent in devices section, if I return to the previous version 1.7 everything is correct

oldmansampson, 2023-12-15:
I've also experienced the same issue. Wanted to add that the Bambu Handy app continues to maintain a solid connection, while Bambu Studio does not. I also did a roll back to the last stable version and everything works as intended.

Mathew-D, 2023-12-15:
It sounds like you guys are having the same trouble I am and reported in  #3045 but mine is with Linux.

Nowi0815, 2023-12-15:
Hi guys,
I have the same problem with a Windows 11 machine.
On my MacBook and on the smartphone, I have no problem.

karlbibby, 2023-12-15:
Same issue after updating today. Bambu handy app stays connected fine so printer must be connected correctly to the network and internet.

Windows 11

critter25, 2023-12-16:
Exact same issue here.  P1S, Windows 10 Pro, camera finally comes on somewhat steadily after actual printing has begun, although it twitches every now and then (i.e. drops sync, then recovers within 10-15 seconds).  Love to see this fixed.  Thanks in advance.

norpol, 2023-12-16:
On Linux I could just help similar with this message:

> Check if your computer has the right time set. I had the same issue and it turned out my network time protocol daemon didn't sync properly.

> https://time.is/ should show max 1-2 seconds or best "your time is exact".

I wonder if they've introduced some time checking in the newer version, would explain why the issue only shows up for  some and even then only on some computers

walterwongbbl, 2023-12-16:
Hi. @x1c3di @mezanu @oldmansampson @Mathew-D @critter25 @karlbibby 
Please follow @norpol   suggestion and synchronize the computer time. We have also identified this issue, and indeed, if the computer system time is not synchronized with the real time, it can cause problems. We will address this problem. For now, please try the following method to synchronize the computer time to the real time and see if it resolves the issue.

Nowi0815, 2023-12-16:
> Hi. @x1c3di @mezanu @oldmansampson @Mathew-D @critter25 @karlbibby Please follow @norpol suggestion and synchronize the computer time. We have also identified this issue, and indeed, if the computer [...]

Thank you, works fine for me!

mezanu, 2023-12-16:
> Hi. @x1c3di @mezanu @oldmansampson @Mathew-D @critter25 @karlbibby Please follow @norpol suggestion and synchronize the computer time. We have also identified this issue, and indeed, if the computer [...]

It has worked for me, the bios time was only one minute less with the windows clock. Thank you!

critter25, 2023-12-17:
> Hi. @x1c3di @mezanu @oldmansampson @Mathew-D @critter25 @karlbibby Please follow @norpol suggestion and synchronize the computer time. We have also identified this issue, and indeed, if the computer [...]

Yep, that seems to have fixed it.  Apparently I was only off by .9 seconds but it's running as before now.  Many thanks!

x1c3di, 2023-12-17:
> Hi. @x1c3di @mezanu @oldmansampson @Mathew-D @critter25 @karlbibby Please follow @norpol suggestion and synchronize the computer time. We have also identified this issue, and indeed, if the computer [...]

Solved it for me as well !, thanks. 
my clock was showing the correct time, so I couldn't figure it out immediately, until I realised I was on the wrong time zone.... duh

oldmansampson, 2023-12-17:
Yup, that solved it. Thanks!

Ursus-T, 2023-12-20:
> Thank you, works fine for me!

It has worked for me too!!!
Thank you!

invadedeluxe, 2023-12-20:
worked for me!

critter25, 2023-12-21:
well, I may have spoken too soon...the camera (eventually) starts, but it is still very intermittent.  Freezes, sits for 10-15 seconds, restarts and can work for several minutes before freezing again.  Cannot find a root cause, all other wi-fi items in the home are working fine with no issues.  <shrug >   At this point it's more of an annoyance than a real problem, but it would be nice to get a steady use from it.

walterwongbbl, 2023-12-21:
Hi. @critter25 
Is it still the same performance as before? 
The cause of this issue is a mismatch between the system time and the server time. We will resolve this problem, and we apologize for any inconvenience caused.

critter25, 2023-12-21:
Not entirely the same.  It is more quickly/consistently opening the camera, but the camera freezes and drops (throws an error) and then retries and reconnects.  There is no real pattern to the drops...it can happen every few minutes or not happen for over 30-40 mins, but that's about the max before it drops.  At no time does it appear to impact the print job, it's just annoying, especially if you're trying to dial in a new filament and hoping to catch failures quickly.  

Again, more of a annoyance but great to hear you're still working on it.  Many thanks for all your efforts.

DankCloudz, 2023-12-21:
I am running the a1 mini and am having the same connectivity issues. I do not see where to "sync" the time between the pc and printer. This is day 3 of not being able to use my printer! Loved my BBLA1M until this!

walterwongbbl, 2023-12-21:
Hi. @DankCloudz 
You need to synchronize your computer's system time because it may have deviated.

Frank833, 2023-12-21:
Same problem here. W11.  Reverting to 1.7 clears it up.  My time is off by about 1 minute from time.gov.  But is sync'd to a Domain Controller so no easy fix.  Having the DC looked at.  But, studio needs to be "fixed" as expecting PC and their cloud servers to be well sync'd to the same clock is not reasonable.

DankCloudz, 2023-12-21:
How do you revert to an earlier FW?

walterwongbbl, 2023-12-22:
@Frank833 @DankCloudz 
In the latest version, in order to maintain connection stability, we have implemented stricter time judgment criteria. However, this has also led to the current issue. If your computer’s system time is not synchronized to standard time or if the time zone settings are incorrect, it can cause the current problem. [We will address this issue with a fix](https://bing.com/search?q=)

frank-family, 2023-12-29:
this is ridiculous had this issue fighting it all day the last 2 and a half days and all it was was a time sync issue lol at least it wasnt something catastrophic thankyou @norpol

GovGeek, 2023-12-31:
Whoa, this worked! Yay!
[image]

Maasi1981, 2024-01-01:
Oh my god, what a relief! I just synced the time now I have stable connection! Thank you!!!

norpol, 2024-02-02:
This appears to have been fixed by https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89 at least #3249 was mentioned to be resolved.

walterwongbbl, 2024-03-11:
This issue will be closed

closed 2024-03-11 by walterwongbbl