bambulab/BambuStudio #3125: prime tower location on buildplate moves to default location after save and reload

JAJordan2515251, 2023-12-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

n/a

### Printer

x1c

### How to reproduce

create and multi color model 
move the prime tower on the build plate to a non default location on the build plate.
save and close the project file
open the project file
prime tower location has moved to the default location

### Actual results

prime tower location has moved to the default location

### Expected results

prime tower location should stay in the location it was saved 

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-16:
@JAJordan2515251 
Hi,

 I have just tried it on windows 10, and can not reproduce your issue
[image]

could you give a short gif for how to reproduce?
and also help to upload the log and 3mf for us debugging

thanks

zhimin-zeng-bambulab, 2023-12-27:
@JAJordan2515251 
I can't reproduce your problem either.  Could you please provide a video of the reproduction?

JAJordan2515251, 2023-12-27:
I’ll need to close the bug report.  It happened to me for two straight days , lots of testing on two different machines, And then on the 3rd day it stopped and I can’t reproduce it anymore eather.

Sorry for raising an alarm and thank you for looking into it.

Jason

zhimin-zeng-bambulab, 2024-01-03:
It's okay. If you have any problems in the future, please feel free to report them.

closed 2024-01-03 by zhimin-zeng-bambulab