bambulab/BambuStudio #3126: BambuStudio doesn't start on MX Linux

Newklolz, 2023-12-15:
### Bambu Studio Version

01.08.01.57 and previous beta versions

### Where is the application from?

Bambu Lab github releases

### OS version

MX Linux

### Additional system information

DELL Precision 5530 with NVIDIA Quadro P1000

[CODE]Graphics:
  Device-1: Intel CoffeeLake-H GT2 [UHD Graphics 630] vendor: Dell driver: i915 v: kernel
    arch: Gen-9.5 process: Intel 14nm built: 2016-20 ports: active: DP-4 off: eDP-1 empty: DP-1,
    DP-2, DP-3, DP-5, DP-6 bus-ID: 00:02.0 chip-ID: 8086:3e9b class-ID: 0300
  Device-2: NVIDIA GP107GLM [Quadro P1000 Mobile] vendor: Dell driver: nvidia v: 525.147.05
    non-free: 530.xx+ status: current (as of 2023-03) arch: Pascal code: GP10x process: TSMC 16nm
    built: 2016-21 pcie: gen: 1 speed: 2.5 GT/s lanes: 16 link-max: gen: 3 speed: 8 GT/s
    bus-ID: 01:00.0 chip-ID: 10de:1cbb class-ID: 0302
  Device-3: Realtek Integrated_Webcam_HD type: USB driver: uvcvideo bus-ID: 1-12:4
    chip-ID: 0bda:5657 class-ID: 0e02 serial: <filter>
  Device-4: Lenovo 500 RGB Camera type: USB driver: uvcvideo bus-ID: 3-2.3.3.3.1:12
    chip-ID: 17ef:482f class-ID: 0e02 serial: <filter>
  Display: x11 server: X.Org v: 1.21.1.7 compositor: xfwm v: 4.18.0 driver: X:
    loaded: modesetting,nvidia unloaded: fbdev,nouveau,vesa alternate: nv dri: iris gpu: i915
    display-ID: :0.0 screens: 1
  Screen-1: 0 s-res: 5120x1440 s-dpi: 96 s-size: 1355x381mm (53.35x15.00")
    s-diag: 1408mm (55.42")
  Monitor-1: DP-4 mapped: DP-1-1 pos: primary model: Samsung C49RG9x serial: <filter> built: 2018
    res: 5120x1440 hz: 60 dpi: 109 gamma: 1.2 size: 1190x340mm (46.85x13.39") diag: 1239mm (48.8")
    modes: max: 3840x1080 min: 720x400
  Monitor-2: eDP-1 note: disabled model: Sharp 0x149a built: 2018 res: 5120x1440 dpi: 142
    gamma: 1.2 size: 344x194mm (13.54x7.64") diag: 395mm (15.5") ratio: 16:9 modes: 1920x1080
  API: OpenGL v: 4.6 Mesa 23.1.2-1~mx23ahs renderer: Mesa Intel UHD Graphics 630 (CFL GT2)
    direct-render: Yes[/CODE]

### Printer

It's not even shipped.

### How to reproduce

1. Download newest Ubuntu AppImage.
2. Make it executable.
3. Run it in the Terminal.

### Actual results

It shows some ssl certificate screen and after that nothing happens. 

$ ./BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage 
[2023-12-15 17:49:27.145210] [0x00007f4f56aabd40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/.../.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/.../.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.376: gtk_window_resize: assertion 'width > 0' failed

(bambu-studio:33163): GLib-GObject-WARNING **: 17:49:33.409: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.409: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:33163): GLib-GObject-WARNING **: 17:49:33.410: invalid cast from 'wxPizza' to 'GtkCellLayout'
[... 148 lines cut ...]
(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.719: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.719: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.726: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.726: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.732: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.732: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.739: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.739: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.746: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:33163): Gtk-CRITICAL **: 17:49:33.746: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault

### Checklist of files to include

- [ ] Log file
- [ ] Project file

n0a110w, 2023-12-16:
I resolved a similar segfault on my MXLINUX installation by blowing away the ~/.config/BambuStudio directory and relaunching

Newklolz, 2023-12-16:
> I resolved a similar segfault on my MXLINUX installation by blowing away the ~/.config/BambuStudio directory and relaunching. try rm -rf ~/.config/BambuStudio 

Thank you! But sadly this doesn't work for me. I can even reproduce the issue on a virtual machine I've just installed. It doesn't have the directory and no other software.

odious, 2023-12-16:
I'm experiencing the same issue on Arch Linux, with both AppImages (Ubuntu and Fedora). 
The Application crashes directly after confirmation of the SSL popup, before any UI is shown.

Stacktrace from gdb:

> Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
> 0x0000555556fa98b1 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()

> (gdb) bt
> #0  0x0000555556fa98b1 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
> #1  0x0000555556f4d0a5 in Slic3r::GUI::StatusBasePanel::create_monitoring_page() ()
> #2  0x0000555556f51414 in Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
> #3  0x0000555556f5c02a in Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
> #4  0x0000555556aaf1ef in Slic3r::GUI::MonitorPanel::init_tabpanel() ()
> #5  0x0000555556aaf984 in Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) ()
> #6  0x00005555568bbd65 in Slic3r::GUI::MainFrame::init_tabpanel() ()
> #7  0x00005555568bd3ee in Slic3r::GUI::MainFrame::MainFrame() ()
> #8  0x0000555556879224 in Slic3r::GUI::GUI_App::on_init_inner() ()
> #9  0x000055555687a26a in Slic3r::GUI::GUI_App::OnInit() ()
> #10 0x0000555559182b97 in wxEntry(int&, wchar_t**) ()
> #11 0x000055555683e68b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
> #12 0x0000555555f05b8a in Slic3r::CLI::run(int, char**) ()
> #13 0x0000555555e259b9 in main ()

Newklolz, 2023-12-17:
> I'm experiencing the same issue on Arch Linux, with both AppImages (Ubuntu and Fedora). The Application crashes directly after confirmation of the SSL popup, before any UI is shown.
>
> Stacktrace from [...]

My output looks similar on MX Linux!

`Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000555556fa98b1 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
(gdb) bt
#0  0x0000555556fa98b1 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
#1  0x0000555556f4d0a5 in Slic3r::GUI::StatusBasePanel::create_monitoring_page() ()
#2  0x0000555556f51414 in Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
#3  0x0000555556f5c02a in Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
#4  0x0000555556aaf1ef in Slic3r::GUI::MonitorPanel::init_tabpanel() ()
#5  0x0000555556aaf984 in Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) ()
#6  0x00005555568bbd65 in Slic3r::GUI::MainFrame::init_tabpanel() ()
#7  0x00005555568bd3ee in Slic3r::GUI::MainFrame::MainFrame() ()
#8  0x0000555556879224 in Slic3r::GUI::GUI_App::on_init_inner() ()
#9  0x000055555687a26a in Slic3r::GUI::GUI_App::OnInit() ()
#10 0x0000555559182b97 in wxEntry(int&, wchar_t**) ()
#11 0x000055555683e68b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#12 0x0000555555f05b8a in Slic3r::CLI::run(int, char**) ()
#13 0x0000555555e259b9 in main ()
`

n0a110w, 2023-12-17:
I replicated the segfault issue in a new MX Linux 23 VM. 
However I don't have the issue on my other MX Linux 23 machine. I'm looking more into it. 
It may help to capture the libraries being called when it segfaults- 
`export LD_DEBUG=libs` and run the AppImage again.

bambu123, 2023-12-18:
Do you have gstreamer installed?

Newklolz, 2023-12-18:
> I replicated the segfault issue in a new MX Linux 23 VM.
> However I don't have the issue on my other baremetal MX Linux 23 machine. I'm looking more into it.
> It may help to capture the libraries being called when it segfaults-
> `export LD_DEBUG=libs` and run the AppImage again.

I don't know if I did it correctly. Let me know if I need to do this in another way. 
[lib.log](https://github.com/bambulab/BambuStudio/files/13706320/lib.log)

> Do you have gstreamer installed?

No, should I install it for a test?

bambu123, 2023-12-19:
> I don't know if I did it correctly. Let me know if I need to do this in another way. [lib.log](https://github.com/bambulab/BambuStudio/files/13706320/lib.log)
>
>
> No, should I install it for a test?

Yes, you should install gstreamer for liveview.

Newklolz, 2023-12-19:
> Yes, you should install gstreamer for liveview.

Wow! I didn't know that.

After running...

`apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio`

...it works in my VM now! Thank you!

But let me ask you one noob question. Should an appimage not have all necessary libs in one package? Kind of confusing to me.

Newklolz, 2023-12-19:
I was able to narrow it down to

`apt install gstreamer1.0-x `

bambu123, 2023-12-19:
> Wow! I didn't know that.
>
> After running...
>
> `apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good [...]

Yes, But we don't have much energy to maintain the Linux platform.

Newklolz, 2023-12-19:
> Yes, But we don't have much energy to maintain the Linux platform.

Thank you for providing a linux version at all :)

closed 2023-12-19 by Newklolz

JoeleJB, 2024-01-15:
> But let me ask you one noob question. Should an appimage not have all necessary libs in one package? Kind of confusing to me.

Yes it should, it is a poor appimage.. I also use MX Linux and a kind user has made a flatpak to get this working for linux bambu users... Seem to work fine without needing to change anything on you host system (much more future proof and easier)..

https://flathub.org/apps/com.bambulab.BambuStudio