bambulab/BambuStudio #3127: Custom filament option is still not really flexible enough

Penguin-fach, 2023-12-15:
I was really hoping that the latest addition of custom filament presets would enable true custom filaments but not so - users are still required to choose from a list of set filament types and that really doesn't help me because I use a lot of PCTG. It is quite similar to PETG but different enough that its behaviour requires separate settings. Not only is it printed at a higher temperature but its shrinkage, stringing, retraction, inter-layer adhesion, cooling requirements and optimum printing speed are all different from PETG.

I would like to see either the list of approved filament types expanded to include more of the oddball filament types (PCTG, PLA-Woodfill, PLA-Metalfill etc) or under the filament type option field an 'other' option that gives the user an option to enter their own filament description. Just like their is a checkbox for 'Can't find vendor I want', maybe their can be a field 'Can't find filament I want'.

Until now I have made do with a very heavily modified PETG profile, but it doesn't display properly and it is all a bit messy and inconvenient.

Maybe under Vendors you could add Fiberlogy - they are quite a big European manufacturer of quality filaments.

DanBao-Bambu, 2023-12-16:
I think I should know your needs. You need to have complete freedom in type selection, or enter the type you want yourself.
However, in our design now, the Type field is used to filter the selection of suitable existing materials. For example, if PLA is selected, ABS material will not appear in the Filament selection box below. This is because creating new custom materials based on the same type of Filament will make modifying parameters more convenient and less error prone.
I think your request to use PCTG and other types is reasonable, but there will be no Filaments of the same type created in the existing Filaments.
I think I can separate this concept, one TYPE is the real Type you want to create, and the other is based on which Type you want to create a new Filament. Of course, in this case, you still need to make special edits to the preset of the new Filament to match the parameters of the material.
Of course, the new changes require internal discussions within the team, but I am willing to push forward with this idea.

> Maybe under Vendors you could add Fiberlogy - they are quite a big European manufacturer of quality filaments.
Of course.

DanBao-Bambu, 2023-12-21:
After discussion, this feature will cause instability in the existing parameter system, as we have a database for Type. If a Type is created that is not in the database, it will have an impact on the parameter system. You can tell us the Type you want to create, and we will add it to the predefined Type for you.

Penguin-fach, 2023-12-22:
Thank you for your reply.

 

At the moment the three I have used, and which are not in your filament list,  are PCTG, PP (polypropylene) and PC-ABS (Polycarbonate-ABS blend). I have listed some others that I think might also be used. Is it possible to add a description field for odd filaments like plastics that are mixed with ESD materials?

 

PCTG

PP

PC-ABS

 

PA-GF  (Glass fibre)

PETG-GF (Glass fibre)

PC-PMMA

PVB

PVDF (rare filament)

PLA-Wood

PLA-Metal

Polycast (not sure what sort of plastic this is but it can be used for making casting models. I have not used it by I know someone who has tried it).

 

Regards, Frazer

QingZhangBambu, 2024-01-24:
#3205

Penguin-fach, 2024-01-25:
OK, if you can add new types, then please add PCTG, PETG-GF, PP (Polypropylene) and PC-ABS

QuantBits, 2025-07-25:
PC-ABS still missing.

[image]