bambulab/BambuStudio #3128: AMS/External Spool Custom Filament Profile Resetting

kayed721, 2023-12-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

1. Go to Device Tab
2. Select a user created custom filament profile using the newly added custom filament profile feature for any AMS slot or external spool
3. Select a color and PA profile
4. click confirm
5. go to any other tab or AMS or external spool slot
6. return to device tab or previous AMS or external spool slot

### Actual results

The selected AMS or custom spool slot has reset itself to show a question mark (?) without color or PA profile defined anymore

### Expected results

Expected behavior is the user defined custom filament to remain selected and not be reset to empty

### Project file & Debug log uploads

no project file required. all projects affected by this bug. Does not cause a crash so no log generated.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

juledk, 2023-12-15:
yesa i got same problem

DanBao-Bambu, 2023-12-16:
@juledk @kayed721 I understand your situation. If possible, please send out the logs for Bambu Studio and the printer. The printer logs may be a bit large, so you can use Google Drive to share the logs. Thank you very much.

OzDruiD, 2023-12-16:
+ following - defintely as similaries to things I was hitting walls with too, but also with the accesbility side of things, naming etc.

juledk, 2023-12-16:
I dont want to upgrade firmvare before bug is fixed, but ill try to get log files off printer.

 

Mvh 

Ole B. Kristensen
Aarestrupvej 65

9520 Skørping 
mobil nr. 40975053

madtv, 2023-12-17:
I have the same issue with the X1C and Bambu Studio 1.8.2.56. An assigned custom filament in the device section will not be saved and returns to a question mark (?) after assigning another AMS slot with a custom filament.

pushc6, 2023-12-17:
I am having the exact same issue on 1.8.2.56

kayed721, 2023-12-18:
> @juledk @kayed721 I understand your situation. If possible, please send out the logs for Bambu Studio and the printer. The printer logs may be a bit large, so you can use Google Drive to share the logs. Thank you very much.

How should I share the logs? Where do I send them? Are there separate log files for bambu studio and the printer itself?

kayed721, 2023-12-18:
Also, The user created filament brands do not show in the edit filament menu for the ams on the printer itself using the touch screen.

DanBao-Bambu, 2023-12-18:
Confirm if you have not entered the filter page on the printer screen and click to enter the "edit filter" page. This is a key step to ensure that your printer can obtain the Filament information. If not, the settings on the Studio Device page will not be recognized by the printer. After a period of time, the printer and Studio soybean milk will display "?". This is not related to whether you have set Filament information for other slots. Please confirm.

kayed721, 2023-12-18:
> @juledk @kayed721 I understand your situation. If possible, please send out the logs for Bambu Studio and the printer. The printer logs may be a bit large, so you can use Google Drive to share the logs. Thank you very much.

My bambu studio and printer logs:
https://drive.google.com/file/d/1GF_kgU64jgJ6WYPDu1Z3eTfIVZrOi8rw/view?usp=sharing

DanBao-Bambu, 2023-12-20:
Do you really need to restart the printer? To my knowledge, simply re-enter the Filament editing page to refresh and obtain the newly created Filament.
After entering the Filament editing page and exiting, a new Filament can be selected on the device interface in Studio, unless a new Filament is created again.

OzDruiD, 2023-12-20:
> Simply re-enter the Filament editing page to refresh and obtain the newly created Filament.
> After entering the Filament editing page and exiting, a new Filament can be selected on the device interface in Studio,

Yes, this is how it works for me. (the workaround\fix for ?)   - I dont restart or rebind the printer.

mateuszdrab, 2023-12-31:
This started happening to me a few days ago.
My custom filaments disappeared from the printer's filament selection menu and now I don't even see them in Bambu Studio.

DanBao-Bambu, 2023-12-31:
@mateuszdrab Are you saying that the Filament you created appears in the printer and can be selected through the screen, but it suddenly disappears in Studio? Can you still find the preset you created, or is it missing from the Custom Filament in the wizard interface? If you don't mind, could you send me your log folder and user folder? Thank you very much.

Naninani, 2024-01-06:
Its not just the External - for me it does that for all ams slots.

mateuszdrab, 2024-01-06:
> @mateuszdrab Are you saying that the Filament you created appears in the printer and can be selected through the screen, but it suddenly disappears in Studio? Can you still find the preset you [...]

Unfortunately it's worse than that.
The filaments disappeared from the printer and were available in Bambu studio for some time. An attempt at setting it via the studio would be successful, but the assignment would reset to a question mark after a few seconds.

A few days later, the filament disappeared from the studio and is no longer present in the custom filaments section. However, when I try to create a new one with the same properties, it says it failed to create it.

Happy to send the logs, I'll gather them later.

mateuszdrab, 2024-01-08:
Hi @DanBao-Bambu

I've got the logs ready - do you have an email address/link to some storage where I can upload it?

FunnyKillar, 2024-01-20:
I had the same problem so far, after updating to version 1.8.4.51 I can't select a custom filament in the AMS.

DanBao-Bambu, 2024-01-21:
@mateuszdrab I'm very sorry, I lost your message, it was my mistake. My email is maosheng.wei@bambulab.com . You can upload your log to me. 
@FunnyKillar What is your specific problem? Can't you find the custom consumables you want to set on the printer? Or did it change to "?" a few seconds after setting up custom consumables on the Studio side? Is it still possible to have this custom consumable on the printer, but it is not available in Studio?

FunnyKillar, 2024-01-21:
@DanBao-Bambu 
I created custom filament in the slicer, but it no longer shows up in the filament settings in the AMS.
The dropdown is initially empty, and when the dropdown opens there are only the pre-built profiles from Bambulab.
the problem with the "?" after a short time I have had it so far.

DanBao-Bambu, 2024-01-22:
@FunnyKillar I need your log folder and User folder to help me analyze the problem. To ensure that the problem can exist in the log, can you create a Filment again to reproduce the problem? You can send it to my email and let me know which Filament has the problem, which can make me more targeted.