bambulab/BambuStudio #3129: Firmvare version 01.07.00.00

juledk, 2023-12-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Printer

x1 carbon + ams

### How to reproduce

after update to Firmvare version 01.07.00.00  i got HMS_0300_0200 error 
down grade to previous version no problems

### Actual results

after update to Firmvare version 01.07.00.00  i got HMS_0300_0200 error 
down grade to previous version no problems

### Expected results

after update to Firmvare version 01.07.00.00  i got HMS_0300_0200 error 
down grade to previous version no problems

### Project file & Debug log uploads

after update to Firmvare version 01.07.00.00  i got HMS_0300_0200 error 
down grade to previous version no problems

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-16:
@juledk 
could you give a full snapshot of this information on studio and printer screen?

thanks

juledk, 2023-12-16:
I don’t have any foto/snap off the error but I’ll try to get log files off printer. 
    Mvh Ole B Kristensen Fra mobil

lanewei120, 2023-12-17:
Hi @juledk 
HMS_0300_0200 is not a full error code
our printer team needs the full error code or a snapshot

anyway, if you can provide the log also, they can debug it

juledk, 2023-12-22:
I dont know why but i tried to upgrade firmvare again and this time no error ???

Maybe a glitsh in the first update.

 

 

Mvh 

Ole B. Kristensen
Aarestrupvej 65

9520 Skørping 
mobil nr. 40975053