bambulab/BambuStudio #3131: unexpected layers in multi color print

reebgogs, 2023-12-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

N/A

### Printer

X1C

### How to reproduce

paint colors on a multi object, preview looks as expected
slice and now there are two incorrectly colored layers in between the objects
[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

I did not include the project file because the model is proprietary. I will provide as much information as possible when requested

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-12-18:
@reebgogs  Could your share your 3mf file?

reebgogs, 2023-12-18:
@zhimin-zeng-bambulab Is there a way to share it securely? It is a proprietary model

zhimin-zeng-bambulab, 2023-12-26:
@reebgogs Could you send the model file to me privately? I will only use it for debugging.
This is my email zhimin.zeng@bambulab.com

reebgogs, 2023-12-26:
Thanks! The model has been shared privately.

Greg Wright
916.221.0016

zhimin-zeng-bambulab, 2024-06-26:
It has been fixed in v01.09.03.50

closed 2024-06-26 by zhimin-zeng-bambulab