bambulab/BambuStudio #3133: Add Gcode Line Pause on horizontal slider

RaphTronic, 2023-12-16:
To facilitate troubleshooting print issues, having the ability to Pause on a position of the horizontal slider that steps through the gcode lines would be extremely useful.

For instance. I often need to pause the printer at specific spots within a layer. Like when dialing in Flow & Overlap for new filaments.

The example below illustrates pausing during the 2nd layer. This allows to both check and dial in the 1st Layer's printing parameters for a new filament as the 2 nd layer isn't fully done yet AND check and dial in the 2nd layer parameters at the same time (speed, overlap, flow). Only being able to Insert a pause at the Layer level unfortunately messes with the filament parameter tuning as the pause in between changes the filament response at the start of the next layer  (not the same temperatures when the 2nd layer resumes thus affecting the tuning process).

Today I have to manually pause the printer AFTER it starts the 2nd layer but BEFORE it completes it. Which requires setting up a timer on the computer based on a vague time estimate, then running to the printer to pause it, but not too soon, not too late. Very inefficient and cumbersome.

Another example is pausing within a layer before a known failure occurs (like on bridges or overhangs). Would allow to resume & witness the failure within seconds. Today, on large and complicated models it can take a long time to print a layer depending on filament & settings. So pausing after the initial printing has completed would greatly help troubleshoooting.

Would be extremely helpful for troubleshooting printing issues, and dialing in several printing parameters.

[image]

zacaracaz, 2024-05-18:
Yes, I would love this too!

jlcd, 2024-07-03:
Would also love that

zacaracaz, 2024-07-30:
I feel like this is a really easy feature to add... And I would have thought more people would want this!

LonelyFarts, 2024-12-03:
This would be fantastic to have. Please programmers, help us out!