bambulab/BambuStudio #3135: Organic tree supports are generated incorrectly when variable layer height is used

GuzziRaz, 2023-12-16:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma 14.2

### Printer

X1C

### How to reproduce

1. Load STL file from https://www.printables.com/model/181440-gandalf-the-hobbit-1977
2. Enable supports: Type **tree(auto)** and Style **Tree Organic**.
3. Slice and preview. No problem so far.
4. Change the model to **variable layer height**. _The enclosed project file is stored at this point)._
5. Slice again and watch in Preview how the support fails to meet the model.

### Actual results

All supports interface points are at wrong heights (far too high). Somehow the variable layer height isn't taken into accound.

### Expected results

Correct support generation for variable layer height.

Note that the option **Independent support layer height** neither triggers nor works around the bug: Regardless of this setting, the supports end up at incorrect heights.

### Project file & Debug log uploads

Here's correct supports, object has fixed layer height:
[image]

Here's the incorrect supports, object has variable layer height. All supports end up too high:
[image]

Project file:
[Gandalf.zip](https://github.com/bambulab/BambuStudio/files/13691085/Gandalf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-12-16:
Known issue. Try another support style for now.

jeffmccollum, 2024-12-14:
Still a problem in 1.10.1.50
[image]
[image]

ddmgh, 2025-07-29:
Can confirm this issue persists with 2.1.1.52.