bambulab/BambuStudio #3136: p1p is allowed to damage itself against the motors when setting position through LED display.

TennisRunner, 2023-12-16:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

I go to the LED screen on my p1p and set it to home.
Then I move it to the left or right most side.
Then move it all the way back using the arrow buttons on the p1p.

### Actual results

It hits against the x/y motors making it damage itself.

### Expected results

It should have the x/y coordinates constrained to not allow it to hit against either of the motors when moving manually.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [X] Project file

TaylorN15, 2023-12-16:
How is this a bug with Bambu Studio?

TennisRunner, 2023-12-16:
> How is this a bug with Bambu Studio?

It probably has the same issue when setting the position through the control interface:
[image]

However, I'm not going to risk damaging my printer further to find out.