bambulab/BambuStudio #3137: 暗色模式下打印件的预览图出现白框 In dark mode, the preview image of the object appears with a white box

huangyiqian, 2023-12-16:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Windows11

### Additional system information

CPU：I7 12700KF
GPU：RTX 3060
RAM：16Gb

### Printer

P1s

### How to reproduce

**Switch to dark mode, there is a bug in the preview image**
**切换到暗色模式时，预览图透明通道显示不正常**

### Actual results

[image]

### Expected results

Hope the transparent channel of the preview image in dark mode displays normally
希望暗色模式下的预览图透明通道显示正常

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BambooBugFixer, 2023-12-18:
Hi, thank you for reporting this. Can you provide us with your model? We'll try to reproduce it.

huangyiqian, 2023-12-18:
Regardless of color, all models will have transparent channel display errors. Of course, if you insist, I will upload the problematic slice file @BambooBugFixer 
[透光浮雕.zip](https://github.com/bambulab/BambuStudio/files/13706941/default.zip)

> Hi, thank you for reporting this. Can you provide us with your model? We'll try to reproduce it.

BambooBugFixer, 2023-12-19:
透明通道显示不正常你是指的图片后面白色的背景还是你的模型部分？

huangyiqian, 2023-12-19:
> 透明通道显示不正常你是指的图片后面白色的背景还是你的模型部分？

指的图片后面的白色背景，按道理说我都开了暗色模式了就没必要用白色背景来凸显白色模型了吧

BambooBugFixer, 2023-12-20:
你是对的，我们会改进的

walterwongbbl, 2024-01-08:
请在最新的版本中验证

walterwongbbl, 2024-03-11:
We have optimized this issue and it will be closed

closed 2024-03-11 by walterwongbbl