bambulab/BambuStudio #3138: Custom filaments not avilable in AMS selection

madtv, 2023-12-16:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

A1 Mini

### How to reproduce

1.) Create new custom filament
2.) Go to the AMS Lite and edit AMS filament
3.) Custom filament not available

### Actual results

A new custom filament can be added -> but in the AMS from A1 mini you can not select the Custom Filament. I have also the X1C and its works with this printer

### Expected results

To be able to do the above steps

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DanBao-Bambu, 2023-12-18:
Hello, may I ask what is your firmware version number? When the firmware version is lower than 01.01.50.01, the firmware does not support custom Filament. Please upgrade. If your firmware version number is higher than 01.01.50.01, please share your Studio log and firmware log for analysis.

madtv, 2023-12-24:
I have 01.01.03.00 installed

DanBao-Bambu, 2023-12-25:
I have carefully examined it, and I may have given you incorrect information before. The current version has not yet adapted this feature.

OXERY, 2024-01-03:
Since the regular AMS offers this function, I was quite disappointed that the custom filaments are not selectable in the AMS Lite of my A1. Are there any plans for when this feature will be added? I print a lot, but not necessarily Bambu filament. An alternative for me would be to put the filament list open source on Github so that any manufacturer can also make entries that you can review.

Proxy5, 2024-01-17:
Hi,

A1 and Ams Lite Custom Filament settings, still working, today the Bambustudio has been updated (I use MAC) and unfortunately the Custom Filament settings on the Device tab have disappeared at AMS settings.
[image]

No longer working here:
[image]

The Custom Filament Profile is still visible during the preparation, but it is no longer transferred to AMS.
[image]

[image]

What is the solution?
Downgrade?

walterwongbbl, 2024-01-18:
@bam-winterfell

bam-winterfell, 2024-01-19:
hi@Proxy5 
[image]
Due to some known firmware issues, the new version of Studio and firmware have temporarily disabled this feature for both P and A series. After fixing the firmware, this feature can be used normally

Proxy5, 2024-01-19:
> hi@Proxy5 [...]

Ok thanks, I'll wait patiently then.

bam-winterfell, 2024-01-22:
hi@Proxy5 
I am deeply sorry for any inconvenience caused to you. Thank you for your patience

Nicthalon, 2024-06-07:
Is there any word on an eta for this to be addressed, as it's been 5 months?

DanBao-Bambu, 2024-06-17:
At present, the latest released software for P series and A series printers should be able to support the operation of custom consumables.

anishp55, 2025-04-16:
any update on this?  i am unable to select a custom filament for my A1M with the AMS lite.

cvvs, 2025-05-28:
It's been just shy of 1.5 years. Any word?

noir-cl, 2025-10-23:
any info on this?

creeve4, 2025-11-10:
Any update on this?