bambulab/BambuStudio #3141: bbs_3mf.hpp:264:30: error: ‘ModelObject’ was not declared in this scope

hadess, 2023-12-16:
### Bambu Studio Version

git

### Where is the application from?

Built from source by yourself or others

### OS version

Flathub GNOME SDK 45

### Additional system information

gcc version 13.2.0 (GCC)

### Printer

None

### How to reproduce

Trying to build the latest BambuStudio release

### Actual results

I don't understand how this could have compiled before:
`Model.hpp` which defines `ModelObject` includes `bbs_3mf.hpp` which uses `ModelObject` before it could defined.

```
[274/471] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Printer/PrinterFileSystem.cpp.o
FAILED: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Printer/PrinterFileSystem.cpp.o 
/run/ccache/bin/c++ -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_NO_LIB -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_NO_LIB -DCURL_STATICLIB -DGLEW_STATIC -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENSSL_CERT_OVERRIDE -DOPENVDB_OPENEXR_STATICLIB -DOPENVDB_STATICLIB -DSLIC3R_CURRENTLY_COMPILING_GUI_MODULE -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK3__ -D__WXGTK__ -DwxDEBUG_LEVEL=0 -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/run/build/BambuStudio/src -I/run/build/BambuStudio/build/src/platform -I/run/build/BambuStudio/src/hidapi/include -I/run/build/BambuStudio/src/slic3r/Utils -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/run/build/BambuStudio/deps/build/destdir/usr/local/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/atk-1.0 -I/usr/include/blkid -I/usr/include/libmount -I/usr/include/sysprof-6 -I/run/build/BambuStudio/deps/build/destdir/usr/local/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/gio-unix-2.0 -I/usr/include/fribidi -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/gstreamer-1.0 -I/run/build/BambuStudio/build/src/libslic3r -I/run/build/BambuStudio/deps/build/destdir/usr/local/include/opencascade -I/run/build/BambuStudio/src/libnest2d/include -I/run/build/BambuStudio/src/miniz -I/run/build/BambuStudio/src/glu-libtess/include -I/run/build/BambuStudio/src/clipper2/Clipper2Lib/include -I/run/build/BambuStudio/src/minilzo -isystem /run/build/BambuStudio/src/eigen -isystem /run/build/BambuStudio/src/libigl -isystem /run/build/BambuStudio/deps/build/destdir/usr/local/lib/wx/include/gtk3-unicode-static-3.2 -isystem /run/build/BambuStudio/deps/build/destdir/usr/local/include/wx-3.2 -isystem /run/build/BambuStudio/deps/build/destdir/usr/local/include -isystem /run/build/BambuStudio/deps/build/destdir/usr/local/include/boost-1_78 -isystem /run/build/BambuStudio/deps/build/destdir/usr/local/include/OpenEXR -std=gnu++20 -fext-numeric-literals -Wall -Wno-reorder -pthread -O3 -DNDEBUG -std=gnu++17 -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -DOPENVDB_ABI_VERSION_NUMBER=8 -MD -MT src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Printer/PrinterFileSystem.cpp.o -MF src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Printer/PrinterFileSystem.cpp.o.d -o src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Printer/PrinterFileSystem.cpp.o -c /run/build/BambuStudio/src/slic3r/GUI/Printer/PrinterFileSystem.cpp
In file included from /run/build/BambuStudio/src/slic3r/GUI/Printer/PrinterFileSystem.cpp:3:
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:264:13: error: variable or field ‘save_object_mesh’ declared void
  264 | extern void save_object_mesh(ModelObject& object);
      |             ^~~~~~~~~~~~~~~~
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:264:30: error: ‘ModelObject’ was not declared in this scope
  264 | extern void save_object_mesh(ModelObject& object);
      |                              ^~~~~~~~~~~
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:264:43: error: ‘object’ was not declared in this scope; did you mean ‘nlohmann::detail::value_t::object’?
  264 | extern void save_object_mesh(ModelObject& object);
      |                                           ^~~~~~
      |                                           nlohmann::detail::value_t::object
In file included from /run/build/BambuStudio/src/nlohmann/detail/exceptions.hpp:8,
                 from /run/build/BambuStudio/src/nlohmann/detail/conversions/from_json.hpp:15,
                 from /run/build/BambuStudio/src/nlohmann/adl_serializer.hpp:6,
                 from /run/build/BambuStudio/src/nlohmann/json.hpp:51,
                 from /run/build/BambuStudio/src/libslic3r/Format/../GCode/ThumbnailData.hpp:6,
                 from /run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:4:
/run/build/BambuStudio/src/nlohmann/detail/value_t.hpp:43:5: note: ‘nlohmann::detail::value_t::object’ declared here
   43 |     object,           ///< object (unordered set of name/value pairs)
      |     ^~~~~~
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:266:13: error: variable or field ‘delete_object_mesh’ declared void
  266 | extern void delete_object_mesh(ModelObject& object);
      |             ^~~~~~~~~~~~~~~~~~
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:266:32: error: ‘ModelObject’ was not declared in this scope
  266 | extern void delete_object_mesh(ModelObject& object);
      |                                ^~~~~~~~~~~
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:266:45: error: ‘object’ was not declared in this scope; did you mean ‘nlohmann::detail::value_t::object’?
  266 | extern void delete_object_mesh(ModelObject& object);
      |                                             ^~~~~~
      |                                             nlohmann::detail::value_t::object
/run/build/BambuStudio/src/nlohmann/detail/value_t.hpp:43:5: note: ‘nlohmann::detail::value_t::object’ declared here
   43 |     object,           ///< object (unordered set of name/value pairs)
      |     ^~~~~~
/run/build/BambuStudio/src/libslic3r/Format/bbs_3mf.hpp:288:32: error: expected ‘)’ before ‘&’ token
  288 |     SaveObjectGaurd(ModelObject& object);
      |                    ~           ^
      |                                )
In file included from /run/build/BambuStudio/src/slic3r/GUI/Printer/PrinterFileSystem.cpp:4:
/run/build/BambuStudio/src/libslic3r/Model.hpp: In member function ‘void Slic3r::ModelObject::load(Archive&)’:
/run/build/BambuStudio/src/libslic3r/Model.hpp:688:21: error: ‘save_object_mesh’ is not a member of ‘Slic3r’
  688 |             Slic3r::save_object_mesh(*this);
      |                     ^~~~~~~~~~~~~~~~
/run/build/BambuStudio/src/libslic3r/Model.hpp: In member function ‘void Slic3r::ModelVolume::load(Archive&)’:
/run/build/BambuStudio/src/libslic3r/Model.hpp:1200:21: error: ‘save_object_mesh’ is not a member of ‘Slic3r’
 1200 |             Slic3r::save_object_mesh(*object);
      |                     ^~~~~~~~~~~~~~~~
```

### Expected results

No compilation errors.

### Project file & Debug log uploads

As above

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-12-21:
@hadess 

as we can not reproduce this issue here

how about add a line here

```
$ git diff
diff --git a/src/libslic3r/Format/bbs_3mf.hpp b/src/libslic3r/Format/bbs_3mf.hpp
index af6994b77..d2f46be3e 100644
--- a/src/libslic3r/Format/bbs_3mf.hpp
+++ b/src/libslic3r/Format/bbs_3mf.hpp
@@ -8,6 +8,7 @@

 namespace Slic3r {
 class Model;
+class ModelObject;
 struct ConfigSubstitutionContext;
 class DynamicPrintConfig;
 class Preset;
```

hadess, 2023-12-21:
> how about add a line here

That's what I did in https://github.com/bambulab/BambuStudio/pull/3139 :)

lanewei120, 2023-12-21:
@hadess 

Hi, I just checked our building on OpenSUSE which has a gcc version 13.2.1
it can also compile successfully

so it is very strange there are so many compiling errors on your side
did you modify some compiling options?

hadess, 2023-12-27:
> Hi, I just checked our building on OpenSUSE which has a gcc version 13.2.1 it can also compile successfully
> 
> so it is very strange there are so many compiling errors on your side did you modify some compiling options?

I believe that 13.2.1 is pre-release, 13.2.0 is the current stable version of gcc. The only change compared to your setup is:
1. Flatpak GNOME SDK instead of SUSE's setup
2. wxWidgets 3.2.4, because the version bundled in BambuStudio doesn't link to EGL properly.

The rest is standard. I don't know what Boost or wxWidgets changes would have caused all the includes problems I saw, but I've been able to build and run BambuStudio with all the changes in https://github.com/bambulab/BambuStudio/pull/3139

Looking at the code, I honestly don't see how it managed to compile before...

closed 2024-01-02 by lanewei120