bambulab/BambuStudio #3142: Option to remove Daily Tips popups

MrTantrum, 2023-12-16:
In the latest Bambu Studio release 1.8.2.56, now tips popup with slicing models and I cannot find a way to disable them.  This "feature" is extremely annoying.  These "tips" are geared towards newbies and not experienced and/or professional 3D makers and are completely unnecessary.  

Is there a way to disable them?  If not, can you PLEASE add a configuration option to disable them in the next Studio release?

robert-tables, 2023-12-16:
Wow, what an unhinged way to ask for this. How can a small popup in the lower corner be "extremely annoying"?

huangyiqian, 2023-12-16:
True.It's a bit annoying, the tips in this version are much larger than before

MrTantrum, 2023-12-16:
I could sugar coat things if you want, but just getting right to the point.  Yes, it is visually extremely annoying and totally unnecessary.  Just as annoying as my other issue that hasn’t been resolved of forcing a warning message about leaving the door open every single time I initiate a print on the X1 from the memory card.  Your users know very well how to use the software and the machine, so these elementary nags are 100% unnecessary.

robert-tables, 2023-12-16:
It's a minor annoyance and doesn't affect anything else whatsoever, yet you're here ranting. Learn to ignore things and grow up.

LJKraus, 2023-12-16:
I don't see a rant.  I see a request to disable "tips" that are useless for any one with any experience and _is_ quite annoying, especially when the screen/window/zoom size means that the popup covers the display of the model. 

I was unhappy to see tips were still present when I first ran v01.08.02.56,  but today I am not seeing them after slicing. I'm not sure what I might have changed @dilwei3374, but there might be hope that they will stop for you as well.  I don't see a way to turn them back on.  Viewing help tips needs to be selectable in Preferences.

MrTantrum, 2023-12-16:
How about design software with features that can be selectable to accommodate all levels of users.  Every time I create a spreadsheet, I don't have Excel telling me with giant popups how to write formulas, etc.  Remember just how popular Clippy was (for those that don't the outrage over it was massive and MS eventually did away with it)?

dilwei3374, 2023-12-16:
> I don't see a rant. I see a request to disable "tips" that are useless for any one with any experience and _is_ quite annoying, especially when the screen/window/zoom size means that the popup [...]

Not sure why I was tagged here but I guess I'll give my 2-cents haha. The window is a little annoying, but if it were smaller I don't think I would care. Being able to disable the window altogether would be a nice option, even if its just because its one more thing you have control over.

psy0rz, 2023-12-17:
It's super annoying, its too big and i have to click to close it every time.

Didn't we learn anything from Clippy?

robert-tables, 2023-12-17:
> How about design software with features that can be selectable to accommodate all levels of users. Every time I create a spreadsheet, I don't have Excel telling me with giant popups how to write [...]

There you go again, using over the top language like "outrage" over a minor annoyance. Sounds like you're the one with the issues, not this project.

Merlwynd, 2023-12-18:
Just an fyi...Orca Slicer (fork of Bambu Studio) has a toggle for activating/deactivating daily tips as an option under Preferences. Should Bambu decide to do this for Bambu Studio, the code is already available from Orca Slicer....

LiZ-Li-BBL, 2023-12-19:
Hello everyone,
The tips can be folded, and once you fold it, it remains folded every time it pops up. And when you want to view tips, you can always expand it again.
[image]
[image]

psy0rz, 2023-12-19:
ah that makes it much better. the problem was it was so big it was obscuring the preview.

for me its ok like that, and i feel like a moron that i didn't discovered that myself :)

thanks

xSepticSidx, 2023-12-27:
I What in BambuStudio.conf does ""show_daily_tips": true," do? 
Should setting it false not show the daily tips? Changing it to false does not appear to do anything?

Sortam, 2023-12-30:
This "daily hints" thing is very annoying! There definitely should be an option to disable it.  
Also, I would like to have an option to keep the state of the "Color scheme" drop-down.  It unwinds every time after the slicing is complete, which is totally unnecessary in most cases. 
The "Slicing finished" notification should be moved somewhere off the workplace - write it in the menu bar area instead of pop-up, before the Share/Slice/Print buttons for example. 
Current UX in this part of the workflow is simply bad: instead of the slicing results the user sees big pop-ups and drop-downs that block half of the model screen and require two clicks every time to remove them. When you need to re-slice your model 10+ times (while setting filament changes for particular layers, for example) this behavior can drive even a totally phlegmatic person like me crazy.

dohlin, 2024-01-07:
> I What in BambuStudio.conf does ""show_daily_tips": true," do? Should setting it false not show the daily tips? Changing it to false does not appear to do anything?

I believe there's a difference between the 'daily tips' that can show on startup (what you're referring to) vs. those that show after slicing.

I'm with everyone else there should be an option to disable the post-slicing tips as well as they don't serve any real valuable purpose.

buffalodragon, 2024-01-10:
To add to this, I would like the option to do away with any hint of tips being done away all together. Even seeing this there irritates me when I'm going through the G-Code preview.
[image]

Please have a toggle in preferences to outright remove the "Daily Tips" feature all together.

dohlin, 2024-01-10:
> To add to this, I would like the option to do away with any hint of tips being done away all together. Even seeing this there irritates me when I'm going through the G-Code preview. [...]

There's already an option to turn off the code preview after slicing...unless you're asking for something different that I'm not understanding?

buffalodragon, 2024-01-11:
> There's already an option to turn off the code preview after slicing...unless you're asking for something different that I'm not understanding?

I am asking for the removal of the "Daily Tips" toggle, as circled within the screenshot.

dohlin, 2024-01-11:
> I am asking for the removal of the "Daily Tips" toggle, as circled within the screenshot.

Gotcha sorry misread that my bad.  Yeah complete removal option is I think what everyone here is wanting lol

LiZ-Li-BBL, 2024-01-16:
We will discuss it.
Thanks!

zlice, 2024-06-06:
I feel like this has gotten worse. Every time I re-slice it comes back, may or may not toggle expand/collapse for tips.

alanLacey44, 2024-06-18:
Agree, I'd like to turn it off too, mostly

ft276, 2024-09-02:
I'd also like to dispose of it if possible please.

fasn8ya, 2025-02-22:
does anyone know how to disable the color scheme pop up from appearing after every time you slice ? its very annoying. you have to click on it to get it out of the way in order to clearly view the model every time

herrdeh, 2025-05-18:
I would also like to politely request the option of being able to see the object immediately after slicing and not having to manually remove any windows or overlays in front of it.