bambulab/BambuStudio #3143: Bambu Studio crashes

struppi1963, 2023-12-16:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro Version 23H2

### Additional system information

Gerätename	Christophs-HP-ProBook
Prozessor	Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz   1.99 GHz
Installierter RAM	16,0 GB (15,9 GB verwendbar)
Geräte-ID	635238D1-C854-4B87-9BF0-A94B4A211ABA
Produkt-ID	00330-52771-40444-AAOEM
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor

### Printer

Bambu Lab P1S

### How to reproduce

A project is running normally and the device side is open so that the progress of processing can be observed via the camera.
The camera keeps crashing, but it comes back online after a short time.
What's worse is that Bambu Studio crashes completely.
These crashes have happened several times now.
When you restart the studio, the software recognizes that there is a running project and reloads it. The connection to the device is also available again.
There were no camera failures or studio crashes in the previous version.

### Actual results

nothing to write

### Expected results

nothing to write

### Project file & Debug log uploads

Studio just runnning

### Checklist of files to include

- [ ] Log file
- [ ] Project file

struppi1963, 2023-12-16:
[debug_Sat_Dec_16_16_30_27_32268.log.zip](https://github.com/bambulab/BambuStudio/files/13693544/debug_Sat_Dec_16_16_30_27_32268.log.zip)
Another crash just now.

bambu123, 2023-12-18:
It seems to be caused by insufficient memory. We are preparing a temporary modified version. Please help test it.

bambu123, 2023-12-18:
@struppi1963
Please help test this temporary modified version.
https://1drv.ms/u/s!AkMfUF_oU9pBnno4A6wYQivh107j?e=a40FIG

struppi1963, 2023-12-18:
I installed this version and tested it for several hours today. There was no crash during operation.
The camera stopped working a few times. The error number was 4. After a short time, the camera reconnects itself. The printer is connected via WLAN with 2.4 GHz and a data rate between 50 - 150 Mbit/s.

admiralBrains, 2023-12-18:
I've also been getting constant camera disconnect/crashing in this version of bambu studio. Usually the camera feed crashes and then reconnects, sometimes it won't work again unless I switch tabs (ie switch to prepare then back to device) other times it won't work again unless I close and reopen bambu studio. I've not had any issues with the app itself crashing. I also had no camera problems with the previous release.

bambu123, 2023-12-19:
> I installed this version and tested it for several hours today. There was no crash during operation. The camera stopped working a few times. The error number was 4. After a short time, the camera [...]

Thank you for your feedback. It seems that the changes are effective and we will merge them into the official version as soon as possible. For issues with Liveview, please help provide Studio's log.

bambu123, 2023-12-19:
> I've also been getting constant camera disconnect/crashing in this version of bambu studio. Usually the camera feed crashes and then reconnects, sometimes it won't work again unless I switch tabs [...]

What is your previous version? Also, please help provide Studio's log.

struppi1963, 2023-12-19:
> Thank you for your feedback. It seems that the changes are effective and we will merge them into the official version as soon as possible. For issues with Liveview, please help provide Studio's log.

Can you please tell me briefly where I can find the log file?

admiralBrains, 2023-12-19:
> What is your previous version? Also, please help provide Studio's log.

Hi - my previous version that I had no issue with was v01.07.07.89
My current version is v01.8.2.56
[log.zip](https://github.com/bambulab/BambuStudio/files/13715852/log.zip)

I'm not sure which log files you need so have attached everything in the log folder - I have printed this morning and afternoon and have been encountering the camera feed issue, had to restart Bambu Studio one time when the camera feed kept failing to start.

admiralBrains, 2023-12-19:
> Can you please tell me briefly where I can find the log file?

Should be in appdata/roaming, try:

C:\Users\**YOURUSERNAME**\AppData\Roaming\BambuStudio\log

struppi1963, 2023-12-19:
https://upload.disroot.org/r/TJ6a1xqH#Px6KAmvWA7Ev8pV4JrY0J5guZCZ7AysSXDIBwtkdc5Q=

bambu123, 2023-12-20:
> Hi - my previous version that I had no issue with was v01.07.07.89 My current version is v01.8.2.56 [log.zip](https://github.com/bambulab/BambuStudio/files/13715852/log.zip)
>
> I'm not sure which log [...]

It seems to be a problem with printer v1.4, and you need to restart the printer to restore it. Upgrading the printer to v1.5 can also solve the problem.

admiralBrains, 2023-12-20:
> > > > I've also been getting constant camera disconnect/crashing in this version of bambu studio. Usually the camera feed crashes and then reconnects, sometimes it won't work again unless I switch tabs (ie switch to prepare then back to device) other times it won't work again unless I close and reopen bambu studio. I've not had any issues with the app itself crashing. I also had no camera problems with the previous release.
> > > 
> > > 
> > > What is your previous version? Also, please help provide Studio's log.
> > 
> > 
> > Hi - my previous version that I had no issue with was v01.07.07.89 My current version is v01.8.2.56 [log.zip](https://github.com/bambulab/BambuStudio/files/13715852/log.zip)
> > I'm not sure which log files you need so have attached everything in the log folder - I have printed this morning and afternoon and have been encountering the camera feed issue, had to restart Bambu Studio one time when the camera feed kept failing to start.
> 
> It seems to be a problem with printer v1.4, and you need to restart the printer to restore it. Upgrading the printer to v1.5 can also solve the problem. The version 1.5 is under testing, please wait.

I do turn it off nightly but understood, I will wait for v1.5, thank you.

matbroomfield, 2023-12-29:
Bambu Lab 1.8.2.56 constantly crashes my computer while multitasking. Carbon X1 + AMS. 64GB ram, threadripper 3960X, ROG Zenith Extreme Alpha 2 motherboard plenty of drive space. Drivers installed except audio. When I had them installed, I kept getting the warning that something had injected incompatible DLLs. Deleted the audio drivers which resolved the problem at the cost of 3D audio. Whenever I click to connect Studio to my printer for a live view, I get the sound that usually means a USB device is connecting. The cursor freezes and the screen goes black as through graphics drivers are installing. The screen returns. The cursor shrinks then returns to 5x size. The print begins as normal, but if I am playing a game with Studio running in the background (which I have been able to do just fine all year), the computer just randomly crashes. after anything from 30-60 minutes.
Any thoughts please devs?

bambu123, 2023-12-29:
> Bambu Lab 1.8.2.56 constantly crashes my computer while multitasking. Carbon X1 + AMS. 64GB ram, threadripper 3960X, ROG Zenith Extreme Alpha 2 motherboard plenty of drive space. Drivers installed [...]

Please follow the wiki and config GPU for BambuStudio.
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

SaltWei, 2024-01-18:
> Please follow the wiki and config GPU for BambuStudio.
> https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

@matbroomfield Is this helpful to solve the problem you met?

closed 2024-08-06 by SaltWei