bambulab/BambuStudio #3144: Bambu Studio Freezes when editing AMS filament settings

Kokorone, 2023-12-16:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2 

### Additional system information

12700K, RTX 4080, 32GB RAM, SSD Storage

### Printer

Bambu Lab P1S

### How to reproduce

1. Open Bambu Studio
2. Navigate to Device tab
3. Observe AMS filament settings on the right side of window
4. Click on pencil icon on any of the 4 AMS slots to edit filament (color, flow rate, etc)

### Actual results

1. Bambu Studio will silently freeze
2. Click actions from this point will generate a generic windows error sound
3. Bambu Studio will need to be force quit via Task Manager at this point as it will not recover or respond

### Expected results

AMS settings for specified filament should open and be editable

### Project file & Debug log uploads

[Bambu P1S Poop Chute.zip](https://github.com/bambulab/BambuStudio/files/13693892/Bambu.P1S.Poop.Chute.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/13693893/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bam-winterfell, 2023-12-21:
@Kokorone When you have finished editing the AMS material, you can click these two buttons to close the window. 
[image]
I'm not sure if this has solved your problem. If not, please continue to ask.

Kokorone, 2023-12-21:
> @Kokorone When you have finished editing the AMS material, you can click these two buttons to close the window. ![58ee8902-316f-4b97-b11f-6792a26d9ce0](https://private-user-images.githubusercontent.com/142556838/292072878-b035c844-81f4-45b8-bb0a-f52cb384bb65.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxMjc0NTcsIm5iZiI6MTcwMzEyNzE1NywicGF0aCI6Ii8xNDI1NTY4MzgvMjkyMDcyODc4LWIwMzVjODQ0LTgxZjQtNDViOC1iYjBhLWY1MmNiMzg0YmI2NS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjFUMDI1MjM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGJiZDg5NGUxYzg5YjhkNjZlYWFlNTQyZjU4ZWUyYmZlNWViNmMyNmJiZjBmNTk5YmNiOTNhZmI4N2Q0ZmE2YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.QoHCkfzbBNURtoTOi_CIwbivq1J_WvbSNlkqpaE9maU) I'm not sure if this has solved your problem. If not, please continue to ask.

That menu you are showing there never even opens for me. Trying to open that menu crashes and freezes bambu studio

bam-winterfell, 2023-12-25:
Hi@Kokorone
Would you please provide a video or picture of the operation process? This will help me locate the problem. Thank you

KiroDragon, 2023-12-31:
We are having the same issue I think 
Whenever we try to change a setting of a filament and safe it as a profile (new or existing one) the program crashes/freezes 

Did a video but couldn't compress it into 10mb so here's my Dropbox link https://www.dropbox.com/scl/fi/n9n08mteg1h8rmyuwbo0d/20231231_192806.mp4?rlkey=kmml7w6m91fmlmf0qkoh5tsm5&dl=0 

Currently using: 

Bambu Studio Version
1.8.2.56

Where is the application from?
Bambu Lab Official website

OS version
Windows 11 Pro 22H2

Additional system information
i7 1265U, 32GB RAM, SSD Storage

Printer
Bambu Lab X1C

DanBao-Bambu, 2024-01-18:
@Kokorone Hello, I have noticed your issue and it should have been resolved in the latest Studio 1.8. You can try using it and feel free to ask any questions.

Kokorone, 2024-01-18:
@DanBao-Bambu for my issue, I've just updated to 1.8.4 and it still has the issue where bambu studio freezes when attempting to edit an ams filament setting

bam-winterfell, 2024-01-19:
hi@Kokorone 
I'm sorry for any inconvenience caused to you.Would you please provide the log file? The specific location of the log is here:
[image]
Before seeing the log file, I speculate that it may be due to issues with custom material presets. You can try deleting this folder and closing the synchronization preset in BambuStudio, then opening the material editing window again.
[image]
[image]

Kokorone, 2024-01-19:
> hi@Kokorone I'm sorry for any inconvenience caused to you.Would you please provide the log file? The specific location of the log is here: [...]

I'm attaching the 4 latest log files. 
[debug_Thu_Jan_18_11_04_53_41560.log.zip](https://github.com/bambulab/BambuStudio/files/13983345/debug_Thu_Jan_18_11_04_53_41560.log.zip)

I will try and delete the filament custom material presets but I don't really have anything configured so I'm not sure if that is the issue

bam-winterfell, 2024-01-19:
hi@Kokorone 
There seems to be no problem with the log. Could you please refer to the [link](https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue) and provide the file so that we can locate the problem

Kokorone, 2024-01-19:
> hi@Kokorone There seems to be no problem with the log. Could you please refer to the [link](https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue) and provide the file so that we can locate the problem

I only have one DMP file so I'm not sure if the freezes are being recorded. I've attached it.
[bambu-studio.exe.46208.zip](https://github.com/bambulab/BambuStudio/files/13983499/bambu-studio.exe.46208.zip)

The issue may be fixed after I deleted the filament profile folder though. I have launched Bambu Studio 3 times and have been able to successfully edit the filament settings in the AMS menu each time. I will let you know if it occurs again.

bam-winterfell, 2024-01-19:
@Kokorone 
There should be an issue with this preset(Generic PLA @Creality - Silk ). If you still have this preset after synchronizing, please find it in C:\Users\Username\AppData\Roaming\BambuStudio\user\userID\filament and send it to us. So that we can eliminate such problems in the future. Thank you for your cooperation

Kokorone, 2024-01-19:
> @Kokorone There should be an issue with this preset(Generic PLA @creality - Silk ). If you still have this preset after synchronizing, please find it in [...]

When I resynchronize after deleting the filament folder the silk preset does not repopulate in the folder. Here are the files from when I initially deleted the folder though if you'd like to look at it. 
[Generic PLA @Creality - Silk.zip](https://github.com/bambulab/BambuStudio/files/13983634/Generic.PLA.%40Creality.-.Silk.zip)

bam-winterfell, 2024-01-19:
@Kokorone wooo!!Thank you!!

DanBao-Bambu, 2024-01-19:
@Kokorone 
"Generic PLA @ Creativity - Silk" inherits from "Generic PLA @ Creativity", and the program did not find this preset in the log. This may be because you previously used a Creativity printer and later no longer used it (uncheck from the printer selection boot interface). Causing this problem. If you encounter this problem again, you can try checking this printer again.
But I think the freezing of the AMS settings interface may not be due to this reason. Anyway, if you encounter such problems, please feel free to report to us. Thank you very much.

Goodwine, 2024-02-25:
I have gotten myself into this situation by following the Auto-calibration setup. And I think @DanBao-Bambu 's comment helped me understand why.

In my case, I had a filament setup named something like `Handini PLA Plus @ BBL 0.4mm Nozzle`. After running the "Flow Rate" calibration, the BambuSlicer app asks me to save the new filament profile. So I named it `Handini PLA Plus`. And I think this leads to the same issue because the new filaments profiles name is a prefix of the original one, so it creates a circular dependency if I understand correctly the previous post.

I was able to fix my issue by deleting the new filament profile and trying again but this time using a different name.

bumpusd, 2024-03-13:
I am having the exact same issue as OP. I removed all presets for everything. I uninstalled and reinstalled Bambu Studio. The problem persists that when I click the pencil icon to edit the settings of a filament loaded in the AMS I get the Windows error sound and Bambu Studio freezes hard and I have to End Task in Windows Task Manager.
Bambu Studio Version 1.8.4.5.1
[debug_network_Tue_Mar_12_17_22_24.log.zip](https://github.com/bambulab/BambuStudio/files/14580217/debug_network_Tue_Mar_12_17_22_24.log.zip)
I attached a zip of all my logs from today. If there is anything else I can add to help diagnose this issue let me know.

DanBao-Bambu, 2024-03-18:
@bumpusd I'm sorry for the inconvenience caused. I couldn't find any useful information in your log. The reason for this issue is to create a dump file. If your operating system is Windows, when a freeze occurs, you can open Task Manager, right-click on Bambu Studio, and create a dump file. And send it to me. If possible, please let me know the name you entered when the freeze occurred, which may provide us with some clues.
Here you can find a method to create a dump file：
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

bumpusd, 2024-03-19:
> @bumpusd I'm sorry for the inconvenience caused. I couldn't find any useful information in your log. The reason for this issue is to create a dump file. If your operating system is Windows, when a [...]

Here is a link to a Google Drive folder. The dump file was too large to upload here. I included a screen snip of the device page and when I click the icon that I circled in red it causes the freeze.
https://drive.google.com/drive/folders/14tVtklgklSEzFtcdAYdgTFIWf89Lq52c?usp=drive_link

DanBao-Bambu, 2024-03-20:
@bumpusd
> https://drive.google.com/drive/folders/14tVtklgklSEzFtcdAYdgTFIWf89Lq52c?usp=drive_link
Thank you for sharing, but this link requires access permissions. Can you change the permissions so that I can access the file?

bumpusd, 2024-03-20:
> @bumpusd
> 
> > https://drive.google.com/drive/folders/14tVtklgklSEzFtcdAYdgTFIWf89Lq52c?usp=drive_link
> 
> Thank you for sharing, but this link requires access permissions. Can you change the permissions so that I can access the file?

Done.

GregCareHawk256, 2024-07-31:
I am experiencing the same issue.

bumpusd, 2024-08-01:
I don't know if the most recent updates fixed my issue or if it was when I had to power off the machine and disconnect the AMS to open it up and remove some over dried filament that broke inside the unit. I would suggest powering off your printer and unplugging the AMS for a few minutes then put it all back together and make sure you have the most recent updates. If that doesn't work for you then continue to harass customer support.

GregCareHawk256, 2024-08-26:
Thank you, I ensured there is no filament inside and have installed the update today. 1.9.4.60. I restarted the AMS, restarted the P1S. I have tried just 1 filalment, 2 all the way to full.

This issue remains. 

If I click the pencil icon on either filament 1, 2, 3 or 4, all of which are generic PLA the issue occurs. The program freezes.  Windows 11.

GregCareHawk256, 2024-08-26:
I turned off "Auto Sync User Presets" AND ran Bambu Studio on the main monitor - previously I nearly always use my second monitor. Problem did not recur, whereas it happened every time (maybe 25+ tries?) before I made these two changes.

bumpusd, 2024-08-26:
If the update didn't do the trick it was likely because of the "sync user presets" option. Now that it's working I'd try turning that back on just to see if it breaks it again.

gitagogaming, 2024-12-22:
> I turned off "Auto Sync User Presets" AND ran Bambu Studio on the main monitor - previously I nearly always use my second monitor. Problem did not recur, whereas it happened every time (maybe 25+ tries?) before I made these two changes.

I been running into the same issue, it caused me to stop using my printer for about 6 months

I followed this suggestion and turned off autosync of presets.. 
it didnt work.. so then I moved it to a new monitor.. didnt work.. and then finally to my main monitor and then it magically is fine.. 
So I suspect this is some very odd bug to do with main monitor? 

So now Im just going to leave `user preset sync` turned off as I dont even know or see a purpose of it..
Hopefully issue is done.. this

GregCareHawk256, 2024-12-23:
Glad it worked for you. I found this to be quite mystifying and highly frustrating. Main monitor and the auto sync off was the combo I had to do (as noted above).

BlackRaven82, 2025-02-22:
The same problem still occurs in the latest beta version for Ubuntu.