bambulab/BambuStudio #3145: [Features Request] Measurement function

BambuLabDiscord, 2023-12-17:
Would like the measurement function in Bambu Studio.  It will be useful when we scale an object, we can measure specific parts. At the moment need to switch to another application to check the edited scale.

xSepticSidx, 2023-12-27:
TBH I'm surprised that we do not already have this, the ability to measure edge to edge, corner/radius etc. are essential in CAD.
Taking the model outside of Bambu Studio just for verification means that Studio is lacking so I fully support this request as hard as it may be to implement this new feature.

RaphTronic, 2024-01-04:
Indeed, the lack of a measurement tool makes the workflow very cumbersome when scalig parts, when adjusting clearances for parts that joint together, and when double checking a model's correct version / dimensions before printing.