bambulab/BambuStudio #3148: Camera Gets Draker During Prints - Auto Exposure Only Happens Once at Start?

RipperDrone, 2023-12-17:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Win11 x64

### Additional system information

AMD graphics card, i7 processor

### Printer

BambuLab X1C

### How to reproduce

Start print with high max z extension

### Actual results

See camera getting darker with progression of z coordinate. Bright picture at start, completely dark at end

### Expected results

auto-adjust of camera during printing to compensate the lighting conditions being different for different z height

### Project file & Debug log uploads

[Clipboard01.zip](https://github.com/bambulab/BambuStudio/files/13695937/Clipboard01.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

RipperDrone, 2023-12-23:
[image]

RipperDrone, 2023-12-23:
(this is only after 50 layers, when gettiong to 500 it's like a black hole.. camera needs to redo auto-exposure every x layers, otherwise it's impossible to use those images for monitoring how the print goes.. :-(

CCS86, 2024-04-03:
The exposure control really needs work! 

I recently added some additional interior lighting, and to the eye it looks much better:

[image]

But look at this screen grab from Bambu Studio at the same time:

[image]

CCS86, 2024-05-09:
@QingZhangBambu  @sodachen22 

Have you guys made any progress on this? It makes monitoring prints remotely almost impossible sometimes, and it probably reducing spagheti detection effectiveness.

QingZhangBambu, 2024-05-09:
@wineBambu Please check and update

sodachen22, 2024-05-09:
The exposure is fixed when start the printing job, it is for the same exposure when in timelapse recording case.

CCS86, 2024-05-09:
@sodachen22 I can understand why fixed exposure is better for timelapse quality, but maybe the user should at least have some options:

* Option to keep autoexposure active throughout the print. This would be very helpful in cases where a bare, light-colored print bed, like the textured bed, causes an initially darker exposure setting. But, then after printing a large amount of dark filament, the exposure could be increased.

* Option for an "adder" on the auto-exposure calculation. Gopros use this functionality, where you still have auto-exposure functioning, but the users adds +1, +2, -1, -2, etc to the exposure.

Just look at the histograms of the two images I posted above:

[image]

[image]

There is no room to argue. It is terribly under-exposed.

CCS86, 2024-05-21:
It's not just the gold build plate either:

[image]

CCS86, 2024-06-10:
@sodachen22  @wineBambu

gaolegao-lx, 2024-06-10:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

CCS86, 2024-06-19:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

@gaolegao-lx 

US240619049001

CCS86, 2024-07-06:
@gaolegao-lx 

Were you able to advance this issue?

CCS86, 2024-07-28:
@gaolegao-lx @sodachen22 @QingZhangBambu 

Can we please get an update on this?

CCS86, 2024-09-01:
@gaolegao-lx @sodachen22 @QingZhangBambu

Please!

CCS86, 2024-09-14:
@gaolegao-lx @sodachen22 @QingZhangBambu

Really disappointing that this issue is being ignored.

RipperDrone, 2024-09-21:
Hello Bambu team: This is really frsutrating - look what higher objects look like due to camera exposure still NOT being adjustable to different light conditions dynamically during the print, via a user switch 'dynamic brightness adjustment' yes/no:
[image]

CCS86, 2024-10-05:
@gaolegao-lx @sodachen22 @QingZhangBambu

I have now replaced the camera and the AP board and the issue persists. We need an update on this!

TheRealHotFix, 2024-10-09:
I would like to add that I have an issue with no ability to dynamically adjust the brightness during print, but my issue is my video gets washed out after the print starts.
[image]

The replacement lights I use plug into the same port on the X1C as the original light bar, so the printer turns them on when a print starts with the time lapse enabled, so I have no idea why it doesn't auto-calibrate correctly at the start of the print. I.E. There should be no reason why my video should be washed out like this, and the ability to adjust it after the print starts is now critical for me.

What's astounding to me about this "design decision" is that if you have a room with windows and start a 24 hour print, it's going to get lighter and darker throughout the print process, so why wouldn't you want the camera to periodically adjust to the current brightness levels?

Please allow us the ability to either enable dynamic exposure updates or control it ourselves after a print starts.

TheRealHotFix, 2024-10-11:
Has anyone with an X1C tried flipping the live streaming video from 1080P to 720P? I was having the issue I showed above and I tried changing the resolution from 1080p to 720p and that forced it to re-adjust the brightness. I was then able to flip the video back to 1080p and it appeared to stick?

This seems like a very crude way to do this, and it may only work for X1C printers, but it's better than nothing. I still don't understand why the exposure is off so badlly (super bright in my case or super dim in the case of most everyone else) if the printer is doing a proper auto-adjustment when the print starts.

RipperDrone, 2024-10-11:
> Has anyone with an X1C tried flipping the live streaming video from 1080P to 720P?

Seems like some users have tried it with success. When I tried it, it made my print crash :-(

Shame on Bambu to NOT offer a solution for an issue reported for such a long time and which absolutely seems fixable by offering us users the CHOICE of doing either a one-time only exposure OR a continuous exposure adjustment, e.g. every n layers or even better every x% of printing progress

TheRealHotFix, 2024-10-12:
> Seems like some users have tried it with success. When I tried it, it made my print crash :-(
>
> Shame on Bambu to NOT offer a solution for an issue reported for such a long time and which absolutely [...]

Can you please describe what you men by "made my print crash"? I could see how that might make Bambu Studio crash, but it shouldn't cause the actualy object printing process to stop.

RipperDrone, 2024-10-13:
Didn't expect it to happen either, but the print process froze. I had swapped resolution a few times before and it didn't happen then, so it seems to be an edge case. Maybe buffer overflow or CPU overload processing camera exposure algo and printing g-codes simultaneously.

Hammerfest, 2025-01-20:
Yep, Have the issue here as well with multiple printers, stock light for all, tried a flash light as well without going 3rd party lighting, same issue.

Should be trivial to add toggle for auto focus/auto exposure and maybe a +3 to -3 value selector.
After 2 years (seeing other reports elsewhere), probably not going to be fixed anytime soon.

CCS86, 2025-01-20:
Yup, it's super frustrating that the Bambu team acted as if they were going to dig into this issue, asked for additional information, and then completely ghosted us and won't even reply.