bambulab/BambuStudio #3150: Unable to connect

Pheelgood001, 2023-12-17:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Microsoft Windows [Version 10.0.22631.2861]

### Additional system information

I replaced my PC and am unable to connect with the new one.  Handy still works fine, no network changes on my side.

### Printer

x1c

### How to reproduce

log out, log in, select sync.  No device shows in BS.  I am able to ping the printer by IP address

### Actual results

No device in BS

### Expected results

I should be able to see my damn printer

### Project file & Debug log uploads

Not related to printing

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-12-18:
Hi .@Pheelgood001 
Does it mean that the printer device cannot be searched?

walterwongbbl, 2023-12-18:
Please provide the log of the studio

Pheelgood001, 2023-12-18:
Michael Brierley

415.488.3060

*Note: my email address has changed to ***@***.***"
Thank you and please update your information.

walterwongbbl, 2024-01-08:
@Pheelgood001 
Please provide the log of the studio

Pheelgood001, 2024-01-08:
Already did

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

walterwongbbl, 2024-01-09:
@Pheelgood001 
Can you send the file here? I am unable to receive external emails. Or send it to my email.
https://github.com/bambulab/BambuStudio/issues/3150

closed 2024-03-11 by walterwongbbl