bambulab/BambuStudio #3151: Is there a better way to handle max volumetric flow for ObXidian hot ends?

sprior, 2023-12-17:
Currently the max volumetric flow is set per filament preset.  The machine definition does not have this concept.

This means that if someone is using a E3D ObXidian hot end they'd have to create a whole new set of custom filament presets (since the ObXidian was made in collaboration with Bambu Labs are new presets going to be supplied?)  Also is the firmware for the X1C going to be tweaked to add a HF nozzle option to the nozzle selection in the maintenance screen?

This isn't a great workflow, especially when using Bambu RFID tagged filament because the user would have to override this after syncing with the AMS in BS.

So it would seem that just as you select a different printer config based on your nozzle size that a HF nozzle be treated as a distinct nozzle config, but since volumetric capability isn't currently supported in the printer config that won't work.  And even if it did what would be the interaction between the machine volumetric capability and the filament volumetric flow setting?

I don't know the answer here but wanted to open this to start a discussion.

jpringle03, 2024-01-03:
I agree, I'd love to hear what the answer to this is as it seems like a pain to have to keep changing the settings over and over again after every AMS sync.

sprior, 2024-01-03:
Even if the user is going to create all their own filament profiles Bambu Studio doesn't expose the profile dependency system in their UI so there's no way to filter only the HF profiles to a printer profile with a HF nozzle in a way similar to what BS does for the 0.2 nozzles.