bambulab/BambuStudio #3153: Split to parts misses one of the parts

ctadlock, 2023-12-17:
### Bambu Studio Version

1.8.2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

AMD Ryzen, 16GB

### Printer

Bambu X1C and P1S

### How to reproduce

1. Create a multi body model; in my example I have a base model "spacer" with embossed text that has been filled in to with another body "text".
2. In the slicer, "split to parts" or "split to objects"
3. Look at the parts the slicer detected

[image]

### Actual results

The slicer does not detect one of the bodies. This is not an issue with the actual model. Opening the exported stp file directly shows the appropiate solid bodies. The slicer seems to not detect either the first or last one. I have proven this by creating a test where I add another solid body "test" with the same embossed design; in this case it got the text bodies fine but not the new test body.

[image]

[image]

### Expected results

The slicer should detect all solid bodies.

This is a relatively new bug in the last couple releases. I have printed this same part many times before without issue until recently.

### Project file & Debug log uploads

[roller_spacer_10_test.zip](https://github.com/bambulab/BambuStudio/files/13697916/roller_spacer_10_test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ctadlock, 2023-12-17:
Also happens with OrcaSlicer: https://github.com/SoftFever/OrcaSlicer/issues/3169

EricCraigen, 2023-12-18:
I have been having the same experience with my multi-body models as well..

ctadlock, 2023-12-20:
@QingZhangBambu please advise on this bug, it broke our entire production line

QingZhangBambu, 2023-12-20:
@ctadlock i will talk to the concerned staff of that.

MklBambu, 2023-12-29:
@ctadlock You can try converting STP files to STL files, or try saving them again using 3D software like FreeCAD. I hope this can solve your problem.

ctadlock, 2024-01-03:
> @ctadlock You can try converting STP files to STL files, or try saving them again using 3D software like FreeCAD. I hope this can solve your problem.

Ive been using STP files for over 2 years without issue. Whatever this bug is was introduced recentley. Nothing to do with the output format.

MklBambu, 2024-01-04:
> > @ctadlock You can try converting STP files to STL files, or try saving them again using 3D software like FreeCAD. I hope this can solve your problem.
> 
> Ive been using STP files for over 2 years without issue. Whatever this bug is was introduced recentley. Nothing to do with the output format.

Could you tell me which version is working properly?
thanks

ctadlock, 2024-01-04:
> Could you tell me which version is working properly? thanks

Had to do a bunch of testing and pull out old files. I tested back to v1.6.2 which I know it was working then. Results:

- STL always works
- My old STP file works
- My new STP file does not work

There is very little design difference between the old and new, certainly not a difference in how I emboss the text.

Ive attached my test files in the zip.

So it would seem one of these is true:

1. Inventor changed something about their STP export. Seems unlikely as Ive uploaded/opened these STP files in several 3D viewers and they all look fine.
2. Something about the design of the new STP triggers this bug. No idea what this would be
3. There is a slicer bug where it works with the old STP and not with the new STP. 

If you have a dev env it would take a few min to step through the slicer with the "new.stp" file and look at the logic for splitting to parts.

[image]

[test.zip](https://github.com/bambulab/BambuStudio/files/13827066/test.zip)

MklBambu, 2024-01-22:
hi, I'm sorry for not replying for so long, This is an issue with loading the step file. We are using the OCCT library to load the step file. It is possible that some fields were changed by Inventor, resulting in the OCCT library not being able to read them. We will work hard to fix this issue. Before that, you can try saving it as a step file using other software and try again.

ctadlock, 2025-03-28:
Confirmed this is resolved.