bambulab/BambuStudio #3158: Pause at print sequence by object

Regis-Frollo, 2023-12-18:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab P1S

### How to reproduce

Try to set pauses. 

### Actual results

When selecting print sequence by object it is not possible to add pauses.

### Expected results

Let add pauses even when print by object.

### Project file & Debug log uploads

[no_pause_possible.zip](https://github.com/bambulab/BambuStudio/files/13701170/no_pause_possible.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BambooBugFixer, 2023-12-19:
Hi, thank yo for reporting this. You can actually pause by adding custom gcode between objects. 
[image]

TennisRunner, 2023-12-20:
This should be a feature request. If you can pause on a specific layer in by layer mode. You should be able to pause on a specific layer in by object mode.

QingZhangBambu, 2023-12-20:
That make sense.

mattydboom, 2024-03-26:
Can I ask what the latest is on this feature please, is it a work in progress?