bambulab/BambuStudio #3161: Timelapse smooth travel always present in slicing - no chance to deactivate it

HorstBaumann, 2023-12-18:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

core x7

### Printer

A1 mini

### How to reproduce

any object i slice, gets the "smooth timelapse" typical paths to the back left corner.
Even when the timelaps setting is on "traditional" or timelapse is deactivated.
bambu studio seems to ignore these settings.

[image]

### Actual results

the printed object has ugly filament loops and the printing time increases

### Expected results

a clean print without unnecessary travels

### Project file & Debug log uploads

the problem is not related to a certain project
[Bowling Trophy.zip](https://github.com/bambulab/BambuStudio/files/13706484/Bowling.Trophy.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bachrach44, 2023-12-19:
The A1 is always going to have to move the print head and bed for reasons explained here: https://wiki.bambulab.com/en/software/bambu-studio/Timelapse

However, it would be nice if we could just disable timelapse entirely. It was fun at first, but after a while it's just a waste of time and filament.

DConcord, 2023-12-27:
+1 - I've had occasional luck when toggling Others > Print Sequence > By Object because that mode is not compatible with timelapse. But toggling back doesn't always keep the mystery/hidden timelapse setting disabled

loohney, 2023-12-31:
+1 - We need an option to entirely disable the timelapse feature.

zhimin-zeng-bambulab, 2024-01-03:
[image]
If you choose traditional timelapse and do not check timelapse when sending print job, this GCode will be skipped when printing.

DConcord, 2024-01-03:
> ![image](https://private-user-images.githubusercontent.com/120697589/293935806-e0dc2b2e-04cc-46ac-847e-3b00d14cc5ae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Right, but the estimated time is still thrown off. Especially on bigger prints

SaltWei, 2024-01-15:
> Right, but the estimated time is still thrown off. Especially on bigger prints

This time error is very small and only several minutes for a 16 hours printing.

Hammerfest, 2024-06-15:
v1.9.2 and its still an issue

such a waste of time, just give us the option to turn it off... such a hassle

zhimin-zeng-bambulab, 2024-06-16:
@Hammerfest  There is a timelapse  switch when sending for printing. If you think it is a waste of time, uncheck timelapse and the gcode to the trash will not be executed.

Hammerfest, 2024-06-16:
I do, but I think your missing the point, on the preview, its displays the movement regardless of option, if your trying to optimize line travel this flys in the face of and defeats that ability

lightspeedlucas, 2024-06-17:
+1 - Note that as a workaround, it seems you can fully erase `Time lapse G-code` in the `Machine gcode` tab in your printer's profile and the preview/timing will then work as we expect.

hutje, 2024-07-21:
+1, just also give the option to make a timelapse without the travel movements. Octoprint does provide it and it is just ok for most people. Now it adds 20% extra total time for the travel movements.

Edya7659, 2024-09-25:
Couldn't you just add "Disable" to the Special Mode/Timelapse menu, and then the movement of the print head to the shooting position would not be calculated (disappeared)? Yes, in that case, there would be NO possibility of enabling Timelapse shooting when sending the model to print. But, on the other hand, it would be possible to get more accurate data on the model's printing time.

[image]

rrodriguez-medicaljustice, 2024-10-21:
I just recently bought the A1 and I was wondering about this.

Please just add an option to include or exclude time lapse travel moves, having the time lapse travel moves show up always sis very confusing. 
******I have just spent a few hours trying to figure out why my print was going to take a long time, and figuring out why all those travel lines.****

Tyson-08L, 2025-11-10:
> +1 - I've had occasional luck when toggling Others > Print Sequence > By Object because that mode is not compatible with timelapse (and disables it somehow?). But toggling back doesn't always keep the mystery/hidden timelapse setting disabled

THIS SOLVED IT FOR ME!!!!! THANK YOU SO MUCH!!!