bambulab/BambuStudio #3162: Plate type is transferred from other projects, overrides previous project setting.

julius-koenig, 2023-12-18:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 22H2

### Printer

BambuLab X1Carbon

### How to reproduce

Please see this video on YouTube for a demonstration: <a href="https://youtu.be/KZqPZSMfDcc" target="_blank">https://youtu.be/KZqPZSMfDcc</a>
The raw video file is also included in the "Project file & Debug log uploads" section.

Steps to reproduce:
1. Open any project and select a plate type (here Textured PEI was chosen) and save it.
2. Open a different project and select a different plate type (here: engineering plate) and save this.
3. Now open the first project again and see that the plate type has changed from the first (textured PEI) to the second plate type (engineering).
4. Change the plate type back to the first one (textured PEI) and save the project again.
5. Open the second project again and observe the same behaviour, the first plate type (textured PEI) has overriden the second one (engineering).

### Actual results

The plate type is not saved by project but rather always overriden by the type selected in the previously opened project.
I have only expierienced this behaviour since updating to Bambu Studio 1.8.2.56.

### Expected results

The plate type should be saved in the project and not be transferred from the previously opened project.
This was the behaviour until I updated to Bambu Studio 1.8.2.56.

### Project file & Debug log uploads

Used project files: [project_files.zip](https://github.com/bambulab/BambuStudio/files/13707408/project_files.zip)
Raw video file: [video_file.zip](https://github.com/bambulab/BambuStudio/files/13707415/video_file.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye01, 2023-12-19:
this is new feature:https://github.com/bambulab/BambuStudio/releases/tag/v01.08.02.56
[image]
If you don't want to use this feature, you can turn it off in your preferences settings

julius-koenig, 2023-12-19:
Ah ok, thanks for the header, I've disabled it and now the behaviour I want is back. That's not really the behaviour I understood from the patch note: I would expect it to only change it when I'm switching between printers, not when I'm switching between projects with the same printer selected. Especially if a project has previously had a different plate type manually selected.

I would rethink this feature, or at least setting it as the default, as it e.g. overrides the engineering plate used for higher temperature filaments with the PEI plate, after setting both in the singular projects, as in my example. As there is no error message when using the PETG-CF with the PEI plate, the unwanted behaviour could co undetected until the automatic plate checking of the printer sets in after starting the print (happend once for me). This leads to an annoying time loss because re-slicing and starting is necessary, even though I had set the plate correctly previously, it was just overridden.

Maybe using the previously selected plate type only for new projects, while still saving the bed type in a project for re-opening it would be better, since I think most people work project-by-project for switching plate and filament types according to the project needs.

I'll go ahead and close this issue anyways, thanks for your help!

closed 2023-12-19 by julius-koenig

Haidiye01, 2023-12-19:
When opening a project, a pop-up dialog asks the user if they want to use the preferred printing board. Is this better?

julius-koenig, 2023-12-19:
With preferred, do you mean the previously selected one?

To be honest I don't really see the use-case for the auto-bed-type feature, as I just want to set the bed type once at the beginning of my project and then not worry about it again afterwards. So the previous behaviour (or now with the feature disabled) is just perfect for me.

A pop-up could also get annoying quickly, especially if  you have a project in which you are using multiple project files with different plate types, or are working on multiple projects at the same time and are switching back and forth.

I would leave the feature in but disable it by default, but that's also just me. Maybe ask/poll some other users if they think the feature is useful and should be the default?

Haidiye01, 2023-12-19:
i mean that switching between projects with the same printer selected, a dialog  pop-ups when checked different bed to ask the user if they want to use the preferred bed , Is this better?

julius-koenig, 2023-12-19:
As I said, I want to set the plate type for a project once at the beginning and then forget about it, so any pop-up when re-opening a previously saved project, asking me if I want to change something I previously explicitly set, would lead to more friction and possibly annoyance for users (at least for me). I would therefore not implement a pop-up.

I would simply change the default settings to have "Auto bed type" off, but what the default settings should be isn't really down to one person I think. I would do a poll or ask around for what the default setting for this should be, since 95% of people will probably not change them, so they should be what most people want.

Haidiye01, 2023-12-19:
Okay, looking forward to better suggestions