bambulab/BambuStudio #3163: Software Connects disconects re conects Endless loop

dmccl7, 2023-12-19:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab X1c

### How to reproduce

open software after the update
Click Device
Device connects then fails to connect then connects then fails to connect.......
re installed software
have shut down restarted PC

### Actual results

Continues to connect and then fail to connect and reconects yet the camera still works

### Expected results

before update there was no issue.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13709704/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-12-19:
Hi. @dmccl7 
Please synchronize your system time
https://github.com/bambulab/BambuStudio/issues/3122

dmccl7, 2023-12-19:
It is set to auto syncSent from Da iPhone

spoonwzd, 2023-12-19:
I'm having this exact same issue - updated to the latest release of Bambu Studio and now Studio continually connects/disconnects from my X1C.

It's not an issue with the X1C WiFi connection as I have a steady ping with no drops, plus the camera video feed continues uninterrupted.

I also upgraded the X1C to the latest 1.7 firmware release during troubleshooting, but it hasn't made a difference to the behaviour.

All fingers point to this being an app issue.

dmccl7, 2023-12-19:
Yes I agree it happened only after I updated the software. X1C was already updated to latest firmware and my other computer can connect just fine on a non latest version of the software. I can also use the handy app on my iPhone without issues.Sent from Da iPhone

walterwongbbl, 2023-12-20:
Hi.   @spoonwzd @dmccl7 
Please take a look at this
https://github.com/bambulab/BambuStudio/issues/3122

dmccl7, 2023-12-20:
> Thanks @walterwongbbl that fixed it. Seems my NTP settings hadn't been working for a while and I was actually 56.5 seconds out of sync. Sorting that has resolved the Bambu Studio connection issues.
> Thanks again!

So I did this and seems my clock was out by 5 mins even tho it’s set to auto sync. But I managed to set it manually to the time on my phone and it fixed the issue thank you for your help.Sent from Da iPhone

walterwongbbl, 2023-12-21:
Thank you for your support. If you have any other questions, please provide feedback on GitHub.

closed 2023-12-21 by walterwongbbl