bambulab/BambuStudio #3164: Cannot scroll through SD card videos

sebborn, 2023-12-19:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11, version 23H2

### Additional system information

Device name	Alienware-2023
Processor	13th Gen Intel(R) Core(TM) i7-13700HX   2.10 GHz
Installed RAM	64.0 GB (63.7 GB usable)
Device ID	20BB9D7C-0473-49D5-B03C-F9A12B84F520
Product ID	00342-21090-74012-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Bambu Lab X1C

### How to reproduce

In Bambu Studio

Goto Device - MicroSD Card - Timelapse

My computer shows 9 full file icons, another 3 are partly shown at the bottom. There is no way to scroll down the displayed files if there are more stored on the SD Card.. This has been the case for at least the last 4 iterations of Bambu Studio

### Actual results

[image]

### Expected results

Expected a scroll bar on the right side of the screen

### Project file & Debug log uploads

Not applicable.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-12-19:
You can use the mouse wheel to scroll.

closed 2024-02-06 by bambu123

sebborn, 2024-02-06:
There is no mouse wheel on a laptop computer and I don't want to have to carry an external mouse around with me. Telling me to use the mouse wheel is not a solution.

bambu123, 2024-02-07:
You should be able to use two-finger swipes instead of the mouse wheel.