bambulab/BambuStudio #3165: Crash on Linux when logging in

ReverantGR, 2023-12-19:
### Bambu Studio Version

1.08.02.56

### Where is the application from?

Bambu Lab github releases

### OS version

OpenSUSE 15.4

### Additional system information

CPU: Ryzen 9 5950x
RAM: 64GB
GPU: AMD RX6600XT

### Printer

Bambu Lab X1-Carbon

### How to reproduce

1) Run appImage
2) Wait for login to finish (was logged in from previous version)
3) Crash 

### Actual results

Application receives SIGSEGV as follows:
Thread 593 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffc26ffd700 (LWP 13101)]
0x00007fffdb4ecd0b in OPENSSL_sk_deep_copy () from /usr/lib64/libcrypto.so.1.1

### Expected results

No crash

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13713960/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ReverantGR, 2023-12-19:
I was running 1.7.4.52. After this version, I see OpenSSL was updated in the appImage, and every version after 1.7.4.52 crashes the same way.

norpol, 2023-12-19:
@lanewei120 is already investigating this as far as I understand

https://github.com/bambulab/BambuStudio/issues/1843#issuecomment-1859951337

lanewei120, 2023-12-20:
@ReverantGR 
before this issue solved
you can use the workaround solution just referenced in https://github.com/bambulab/BambuStudio/issues/2663

ReverantGR, 2023-12-20:
@lanewei120 I can confirm this workaround temporarily fixes the issue until the problem is found and solved, thank you!

closed 2023-12-20 by ReverantGR