bambulab/BambuStudio #3167: Adding the translation 'hms_pl' to e.bambulab.com for the Polish language

KrisMorr, 2023-12-19:
Please add the 'hms_pl' file to the Bambulab server. I already have the rest of the translations prepared for this language. Before it is added in its entirety, we would like to test the coherence of the whole with the group. I am attaching a .zip with 'hms_pl'

**Additional context**
[hms_pl.zip](https://github.com/bambulab/BambuStudio/files/13716551/hms_pl.zip)

[image]

arekm, 2024-01-03:
Pull request is missing.

Bartosz2010, 2024-02-18:
[pl]
Czy coś się ruszyło w sprawie polskiego tłumaczenia? Jak na razie, nie ma ani pozycji języka polskiego w liście języków, ani pliku BambuStudio_pl.po w folderze bbl/i18n.

[en]
Has anything happened regarding the Polish translation? As of now, there is neither a Polish language entry in the language list nor a BambuStudio_pl.po file in the bbl/i18n folder.

closed 2024-04-24 by KrisMorr