bambulab/BambuStudio #3168: Fuzzy skin inside question

QoooQ, 2023-12-19:
### Bambu Studio Version

1.7.7.89-1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 64bit

### Printer

bambulab P1SC

### How to reproduce

0.4nozzle 0.2hight layer  3top 3 bottom 15%infill       use fuzzy skin contour 0.8distance 0.3thickness

     Why is there fuzzy skin  inside?

### Actual results

[image]
[image]

### Expected results

[image]
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13715702/log.zip)
[v2_profile200-test.zip](https://github.com/bambulab/BambuStudio/files/13715703/v2_profile200-test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2024-01-22:
[image]
Because the countor and hole is defined in 2D polygon. At this z height of model, the 2D layer is all contour. 
This should be enhancement and also not easy to be enhanced.

SaltWei, 2024-01-22:
[image]
This is layer without problem.

SaltWei, 2024-01-22:
[image]
Same problem also exist on x-y compensation. See wiki explanation here.