bambulab/BambuStudio #3169: Custom Filament Type Selection

jakefallon59, 2023-12-19:
**Is your feature request related to a problem? Please describe.**
When creating a custom filament, I am limited to the generic material options that are listed in the type column. This list is not inclusive of all polymer types which makes some custom filament profiles difficult or misleading to generate. For instance, if I have a Polypropylene carbon fiber material that I would like to build a custom profile for, my only option is to select from the filament types listed, which don't include anything similar to polypropylene carbon fiber. 

[image]

**Describe the solution you'd like**
Please increase the comprehensiveness or the list, or even better would be to allow for custom text to be used for the material type. This process could be identical to the process used for inputting a vendor name that does not appear on the vendor selection drop-down list. 

[image]

**Describe alternatives you've considered**
The only alternative to this process, as it stands today, is to select a material type like "PLA" that does not accurately represent the custom material being developed and can be misleading or confusing for users. 

**Additional context**
Add any other context or screenshots about the feature request here.

DanBao-Bambu, 2023-12-21:
After discussion, this feature will cause instability in the existing parameter system, as we have a database for Type. If a Type is created that is not in the database, it will have an impact on the parameter system. You can tell us the Type you want to create, and we will add it to the predefined Type for you.

jakefallon59, 2023-12-21:
Thank you, Dan!

Here is a list of the ones that I would like to have added.

PP
PP-CF
PP-GF
PE
PE-CF
EVA
PA-GF
PCTG

Also, is there any way to have "Xtellar" added to the vendor dropdown list
when selecting custom filaments?

[image: image.png]

DanBao-Bambu, 2023-12-22:
Okay, I got it. I want to know how these printing materials work on Bambu's printer, what kind of printing board did you use to print these materials, what is the success rate, and what are the key material parameters you set for the materials, such as "Print temperature", "Volume speed limit", or which type of Filament is more similar in properties to Bambu's?
This will be more conducive for us to add the materials you want to our database and provide good experience for others.

tokamac, 2023-12-22:
You are seriously asking us users to provide complete specs for a filament as common (even if it is a "technical" one) as Polypropylene? I have also a roll of [Formfutura 3D Centaur PP](https://formfutura.com/product/centaur-pp/) and I cannot even make a custom filament for it either.
PS: Here is the [technical datasheet](https://formfutura.com/datasheets/formfutura-tds-centaurpp.pdf) for that one.

jakefallon59, 2023-12-23:
Hey Dan,

These polymers differ significantly from those currently listed in Bambu
Studio, which is why their inclusion could be highly beneficial.

Regarding the visibility of these polymer types on the type list, may I
inquire about Bambu's policy? Will Bambu be deciding for the customer which
polymer types they are allowed to create custom material profiles for? Our
experience with the materials I listed previously has been notably positive
over the years on various printing platforms. Allowing users the
flexibility to create custom filament profiles from a comprehensive range
of polymers could not only enhance their creative possibilities but also
potentially broaden the appeal and utility of Bambu printers.

I understand that maintaining a balance between quality control and
offering a variety of materials can be challenging. However, I believe that
empowering customers with more choices can lead to more innovative uses of
Bambu printers, ultimately benefiting both users and Bambu Lab in the long
run.

Here are some of the current settings I am using below but it is important
to point out that there are many different grades of these materials which
will have different processing settings, Much like the current selection of
PLA, PETG, NYLON, ABS, and other polymers currently on the polymer type
list.

Print Temp Success Rate Build Plate Volume Limit Similar to…
PP 230 High Engineering Plate 13 none
PP-CF 240 High Engineering Plate 14 none
PP-GF 240 High Engineering Plate 14 none
PE 230 High Engineering Plate 13 none
PE-CF 230 High Engineering Plate 13 none
EVA 230 Still Trialing since it is a flexible material Engineering Plate 6
none
PA-GF 255 High Any Plate with PVA glue stick 12 PA-CF
PCTG 260 High Engineering Plate 15 PETG

Looking forward to your thoughts on this.

Best,
Jake

Schokobecher, 2024-01-06:
I'm also printing _Formfutura Centaur PP_ using the Engineering Plate and _Magigoo PP_ as adhesive (since PP only sticks to PP) — so the addition of PP would be appreciated. 

Here are my settings for it (around 60hr print time total, no failures)

Softening Temperature: 120 C
Recommended Nozzle Temperature: 220 - 240 C
Nozzle Temp: 235 C (First and Other layers)
Bed Temp 80 C (First Layer) 75 C (Other layers)
Max Volumetric Speed: 6 mm3/s

tokamac, 2024-01-06:
Thanks! Which settings do you use for the cooling section when printing Centaur PP? 
Especially the min and max fans speed and layer time, 
Keep fan always on or not?
Fan speed for overhangs?
and the auxiliary fan speed

Schokobecher, 2024-01-06:
No Cooling for: 1 layers
Min fan speed threshold: 50% / 10s
Max fan speed threshold: 100% / 5s
Keep fan always on: Yes
Slow Printing down: Yes
Min print speed: 20mm/s
Fan Speed for Overhangs: 60%
Aux Fan: Off
Chamber Fan: Off

Profile (from OrcaSlicer):
[FormFutura Centaur PP.json](https://github.com/bambulab/BambuStudio/files/13851316/FormFutura.Centaur.PP.json)

I don't really print things with overhangs in PP, mostly just variantions of the same print in bulk

QingZhangBambu, 2024-01-24:
#3205