bambulab/BambuStudio #3171: When activating the video on BS it warns: Using a BambuSource from a different install,etc.

coat64, 2023-12-19:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 5700G
64 GB DDR4

### Printer

X1C

### How to reproduce

When pressing the play button in Bambu studio. See picture. It is followed by the next error message when pressing Yes.
The possible cause is maybe in that message, it searches in "C:\users\Jan" but the username is longer with spaces and it is cut of after "Jan" where a space is placed. So instead of "C:\users\Jan Further Name" it gets "C:\users\Jan"

After these messages it still works.

I see now that it is probably caused by having BS opened 2times

### Actual results

[image]

My previous version was the Beta version (Studio and Firmware).

### Expected results

Obviously no warning :)

### Project file & Debug log uploads

[debug_Tue_Dec_19_18_48_30_4236.log.0.zip](https://github.com/bambulab/BambuStudio/files/13719700/debug_Tue_Dec_19_18_48_30_4236.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-12-20:
That's it, just follow the prompts to re-register.

coat64, 2023-12-20:
How? When i press yes i got the error referring to a wrong folder. It works but the error keeps popping up when i press play in a new instance of BS.

bambu123, 2023-12-20:
> How? When i press yes i got the error referring to a wrong folder. It works but the error keeps popping up when i press play in a new instance of BS.

Sorry, I didn't notice this problem. There are spaces in the file path causing registration failure. I will fix it as soon as possible. 

If you have installed the Beta version and the official version before, this problem will occur. The two versions cannot be mixed. 

If you are sure to use only one version, you can fix the registration location. Currently, you can manually register with powershell and run 'regsvr32 <path-to-dll>'.