bambulab/BambuStudio #3178: Crash when accessing camera in Bambu Studio

StubbornPodBayDoor, 2023-12-20:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

1. Open Bambu Studio
2.  Click on "Play Button" for camera feed
3. Click anywhere on BS window (no response and taskbar disappears)
4. Click again on BS window (windows notices BS in not responding)
5. Wait (it never recovers)

### Actual results

[image]

### Expected results

Be able to see live feed from printer.

### Project file & Debug log uploads

It happens with every print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-12-21:
Please refer https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view
Problem 5 - The program hung when loading the video stream! (Windows Only)

StubbornPodBayDoor, 2023-12-21:
Oops. Sorry. Thanks for the response. I forgot to mention that I had already set the Bambu Studio to run in my integrated graphics processor in the Nvidia control panel.

bambu123, 2023-12-21:
Thank you for your reply. If you have any further questions, please continue to post them.

StubbornPodBayDoor, 2023-12-21:
It still crashes though, so the problem is not yet resolved.

bambu123, 2023-12-21:
> It still crashes though, so the problem is not yet resolved.
> […](#)

Then, could you send me the DMP log following this wiki?
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

jeffjulian, 2024-01-26:
I have having the same issue.

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works