bambulab/BambuStudio #3179: Embedded Text not working

dylansaintjohn, 2023-12-20:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Ventura 13.3.1 (a)

### Additional system information

Bambu studio is not correctly generating text using the Text shape tool. The text does not embed into the shape per the online instructions.

### Printer

Bambu X1C

### How to reproduce

Generate text on a small object and slice it.

### Actual results

Text is raised above the surface. Never embedded or flush

### Expected results

Text generated should 'merge' with the object and embedded depth should be the amount it's deep into the object.

### Project file & Debug log uploads

[Light Switch Cover- Bambu Bug.3mf.zip](https://github.com/bambulab/BambuStudio/files/13728908/Light.Switch.Cover-.Bambu.Bug.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-12-26:
[image]
@dylansaintjohn 
If you want the text to be flush with the surface of the object, you can change the type of text_shape to "modifier".

zhimin-zeng-bambulab, 2024-01-15:
@dylansaintjohn   Did the above method solve your problem?

dylansaintjohn, 2024-01-18:
I believe that solved the problem. Thank you.

Dylan St. John
(404) 993-1861

LucaUrbinati44, 2024-05-17:
I have the same problem. The "modifier" is not working for me. With "modifier" the text does not even appear after slicing.

AdrianGarside, 2024-10-31:
This still doesn't work for me. I can't embed - does nothing - the text is still on top of the surface and nothing is embedded. I can move the text down and make it a modifier but that also does nothing - there's no impact on the slicing output. This is really frustrating.

donhohler, 2024-11-03:
Same here. Nothing suggested above works. Please fix this.

zhimin-zeng-bambulab, 2024-11-04:
Bug reason: The priority of coloring is higher than that of the modifier, so the color modified by the modifier cannot be displayed

AdrianGarside, 2024-11-04:
> Bug reason: The priority of coloring is higher than that of the modifier, so the color modified by the modifier cannot be displayed

Yes, I discovered that when trying the same thing in Orca. On a standalone object it worked fine in Orca but not in Bambu Studio - the latter still messed up every attempt to embed the text where it worked correctly in Orca.

But on the object I actually wanted it on which was painted in layers, the painting overrode the part (regardless of how I did it - modifier, embedded text, new part with cut out of same size) so the embedded text all showed as if it was there but vanished as soon as you sliced it.

L0b0mau, 2024-12-19:
This worked in previous versions but now after updating to 1.10.1.50 it does not work any longer, regardles what I try with the text :/

claudel2006, 2025-02-01:
I saw that bug after opened a new case! I have exactly the same problem with 1.10.2.64

precipice7, 2025-02-26:
I am able to work around this by adding the text first, BEFORE any coloring. Add text, thickness 0 for flush with object, embedded depth to whatever you need, and change the color to your desired choice. THEN paint whatever colors you need. Works on the newest Bambu Studio version (1.10.2.76)

quadlawnmowerman-coder, 2026-03-05:
issue is back again version 2.5.0.66, text will not imbed and slice, work fine on previous version