bambulab/BambuStudio #3182: Tree support - problem with wall loops

mrpiagg, 2023-12-20:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1 Carbon

### How to reproduce

Enable support
Select Tree hybrid (or any other)
Slice model
Tree support wall loops 0 or 1

### Actual results

Tree support has two wall loops and ingnores setting,

### Expected results

To have just one wall loop or what is set.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2023-12-21:
This is designed feature - tree supports with "Hollow" infill pattern (which is the default) will decide how many loops to use at different layers. 1 layer is at least. If extra strength is needed, 2 layers will be used.
The wall count setting is only valid when you set an infill pattern other than hollow and lightning.

closed 2024-03-23 by tangketan

ajgp, 2025-07-08:
I would like to add I have this issue as well, currently using version 2.1.0.59 on Windows and same on Version 2.1.1.52 on Linux (Opensuse)