bambulab/BambuStudio #3183: Bambulab Studio Share button Icon size too large

AlexAlexBB, 2023-12-20:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Microsoft Windows 11 Pro Version	10.0.22621 Build 22621

### Additional system information

Processor	Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz, 2592 Mhz, 6 Core(s), 12 Logical Processor(s)
Installed Physical Memory (RAM)	64.0 GB
Name	Intel(R) UHD Graphics 630
Name	NVIDIA GeForce RTX 2060

### Printer

Bambu Lab P1P

### How to reproduce

Icon size on share button is always too large and overflows the button boundaries
[image]

### Actual results

Icon size on share button is too large and overflows the button boundaries
[image]

### Expected results

Share icon to fit inside button

### Project file & Debug log uploads

I happens on all projects so I don't think it's project specific. I've included the log if that might be of help. 
[tutk_log_avapi_42924.txt](https://github.com/bambulab/BambuStudio/files/13731044/tutk_log_avapi_42924.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

BambooBugFixer, 2023-12-21:
Hi, thank you for reporting this. Have you modified screen resolution when using Bambulab Studio?

HaullingLlama, 2024-09-28:
I see this issue and also at different screen resolutions other buttons looked messed up such as the buttons on the device tab.
[image]

HaullingLlama, 2024-09-30:
I do believe it is a scaling issue but the UI should be robust enough to handle that. I actually program web UI's and understand the frustration of getting these things fixed sometimes.