bambulab/BambuStudio #3184: Linux Build: Crash on "In pixman_region_append_non_o: The expression y1 < y2 was false"

caco3, 2023-12-20:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

P1S

### How to reproduce

1. Go to the device tab
2. enable the camera

### Actual results

free(): invalid pointer
*** BUG ***
In pixman_region_append_non_o: The expression y1 < y2 was false
Set a breakpoint on '_pixman_log_error' to debug

### Expected results

No crash

### Project file & Debug log uploads

[tutk_log_iotc_3548205.txt](https://github.com/bambulab/BambuStudio/files/13731838/tutk_log_iotc_3548205.txt)
[tutk_log_global_3548205.txt](https://github.com/bambulab/BambuStudio/files/13731839/tutk_log_global_3548205.txt)
[tutk_log_avapi_3548205.txt](https://github.com/bambulab/BambuStudio/files/13731840/tutk_log_avapi_3548205.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-21:
@caco3 

Hello, which image are you using?
ubuntu or fedora?

could you also try another app image?

caco3, 2023-12-21:
I had the issue with `BambuStudio_linux_ubuntu_v01.08.00.62-20231123075101.AppImage`.
I now upgraded to the public beta (`BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage `) and the issue no longer can be reproduced, thus I am going to close the issue.

Thanks for your great work!

closed 2023-12-21 by caco3