bambulab/BambuStudio #3185: not connecting to printer

DankCloudz, 2023-12-20:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

A1 mini + AMS

### How to reproduce

Anytime I open the Bambu Studio app on my pc, it will not connect to the printer to send a print. I have confirmed that they are both connected to the internet and on the same network.
My app is also up to date.

### Actual results

I can not send prints.
It says "Failed to connect to the printer" in orange on Device tab when i click the drop down and select the (offline) printer

### Expected results

to be ab le to print

### Project file & Debug log uploads

[tutk_log_avapi_9456.txt](https://github.com/bambulab/BambuStudio/files/13732547/tutk_log_avapi_9456.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DankCloudz, 2023-12-21:
code 6010

DankCloudz, 2023-12-21:
after successful factory reset, i am getting this message after selecting language and region, when trying to generate QR code to login
"Login failed (83968036)"

DankCloudz, 2023-12-21:
it appears my printer has become disconnected from my account and will not log back in, despite having a solid wifi connection
I suspect the new firmware

walterwongbbl, 2023-12-21:
Hi. @DankCloudz 
Please take a look at this, is it the same as the situation you encountered
https://github.com/bambulab/BambuStudio/issues/3122

DankCloudz, 2023-12-21:
> Hi. @DankCloudz Please take a look at this, is it the same as the situation you encountered #3122

Unfortunately that did not fix my issue, nor do I know how to find the place to "sync" the time in my printer settings and/or in Bambu Studio

walterwongbbl, 2023-12-21:
@DankCloudz 
https://pureinfotech.com/sync-clock-windows-10/

DankCloudz, 2023-12-21:
> @DankCloudz https://pureinfotech.com/sync-clock-windows-10/

i just synced the time on my PC and confirmed correct region is selected. I then restarted the printer but it still wont connect
still getting Code: 6010

DankCloudz, 2023-12-21:
i also tried tried restarting my PC, and the printer. I also logged out of my bambu account and back in and still no luck

walterwongbbl, 2023-12-21:
Hi @DankCloudz 
Please provide Studio's log.

DankCloudz, 2023-12-21:
[debug_network_Thu_Dec_21_15_31_14.log.zip](https://github.com/bambulab/BambuStudio/files/13746021/debug_network_Thu_Dec_21_15_31_14.log.zip)

walterwongbbl, 2023-12-22:
Hi. @DankCloudz 

I reviewed your logs, and it appears that you are unable to connect to our servers. This issue is different from the disconnection caused by unsynchronized system time that I mentioned earlier. You can use our network diagnostic tool to verify if you can use our services normally. Please test it and share the results here, and I will assist you in resolving the issue. 

[image]

DankCloudz, 2023-12-22:
Thu Dec 21 20:58:05:test bambulab start...
Thu Dec 21 20:58:05:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.08.02.56&guide_version=00.00.00.01
Thu Dec 21 20:58:05:test bing start...
Thu Dec 21 20:58:05:[test_bing]: url=http://www.bing.com/
Thu Dec 21 20:58:05:test http start...
Thu Dec 21 20:58:05:test storage start...
Thu Dec 21 20:58:05:[test_oss]: url=http://upload-file.bambulab.com
Thu Dec 21 20:58:05:test storage upgrade start...
Thu Dec 21 20:58:05:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Thu Dec 21 20:58:05:test storage download start...
Thu Dec 21 20:58:05:test plugin download start...
Thu Dec 21 20:58:05:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.08.02.00
Thu Dec 21 20:58:06:test http ok
Thu Dec 21 20:58:06:test bambulab ip resolved = 104.18.3.216
Thu Dec 21 20:58:06:test bambulab ok
Thu Dec 21 20:58:06:[test_plugin_download]: downloadurl=https://public-cdn.bambulab.com/upgrade/studio/plugins/01.08.02.01/363255fecb/win_01.08.02.01.zip
Thu Dec 21 20:58:07:test storage upgrade ok
Thu Dec 21 20:58:08:downloading 5%
Thu Dec 21 20:58:08:downloading 10%
Thu Dec 21 20:58:09:downloading 10%
Thu Dec 21 20:58:09:downloading 15%
Thu Dec 21 20:58:09:downloading 20%
Thu Dec 21 20:58:09:downloading 25%
Thu Dec 21 20:58:09:downloading 20%
Thu Dec 21 20:58:09:downloading 30%
Thu Dec 21 20:58:09:downloading 35%
Thu Dec 21 20:58:09:downloading 40%
Thu Dec 21 20:58:10:downloading 45%
Thu Dec 21 20:58:10:downloading 30%
Thu Dec 21 20:58:10:downloading 50%
Thu Dec 21 20:58:10:downloading 55%
Thu Dec 21 20:58:10:downloading 60%
Thu Dec 21 20:58:10:downloading 40%
Thu Dec 21 20:58:10:downloading 65%
Thu Dec 21 20:58:10:downloading 70%
Thu Dec 21 20:58:10:test storage ok
Thu Dec 21 20:58:11:downloading 75%
Thu Dec 21 20:58:11:downloading 80%
Thu Dec 21 20:58:11:downloading 50%
Thu Dec 21 20:58:11:downloading 85%
Thu Dec 21 20:58:11:downloading 90%
Thu Dec 21 20:58:11:downloading 95%
Thu Dec 21 20:58:11:downloading 60%
Thu Dec 21 20:58:11:downloading 100%
Thu Dec 21 20:58:11:test plugin download ok
Thu Dec 21 20:58:12:downloading 70%
Thu Dec 21 20:58:12:downloading 80%
Thu Dec 21 20:58:13:downloading 90%
Thu Dec 21 20:58:13:downloading 100%
Thu Dec 21 20:58:13:test storage download ok
Thu Dec 21 20:58:15:test bing failed
Thu Dec 21 20:58:15:status=0, body=<!doctype html><html lang="en" dir="ltr"><head><meta name="theme-color" content="#4F4F4F" /><meta name="description" content="Bing helps you turn information into action, making it faster and easier to go from searching to doing." /><meta http-equiv="X-UA-Compatible" content="IE=edge" /><meta name="viewport" content="width=device-width, initial-scale=1.0" /><meta property="fb:app_id" content="570810223073062" /><meta property="og:type" content="website" /><meta property="og:title" content="The veil of winter" /><meta property="og:image" content="https://www.bing.com/th?id=OHR.BavarianSolstice_EN-US9111666986_tmb.jpg&amp;rf=" /><meta property="og:image:width" content="1366" /><meta property="og:image:height" content="768" /><meta property="og:url" content="https://www.bing.com/?form=HPFBBK&amp;ssd=20231221_0800&amp;mkt=en-US" /><meta property="og:site_name" content="Bing" /><meta property="og:description" content="Frost-covered birch trees add a graceful touch to " /><title>Bing</title><link rel="icon" href="/sa/simg/favicon-trans-bg-blue-mg-png.png" /><link rel="preload" href="/th?id=OHR.BavarianSolstice_EN-US9111666986_1920x1080.webp&amp;qlt=50" as="image" id="preloadBg"  /><link rel="preload" href="/rp/tlifxqsNyCzxIJnRwtQKuZToQQw.js" as="script"  /><link rel="preconnect" href="https://r.bing.com" /><link rel="preconnect" href="https://r.bing.com" crossorigin/><link rel="dns-prefetch" href="https://r.bing.com" /><link rel="dns-prefetch" href="https://r.bing.com" crossorigin/><style type="text/css">@media(max-width:1440px){#id_n{white-space:nowrap;overflow:hidden;text-overflow:ellipsis;max-width:90px;display:inline-block}}.rwds_svg.hp{margin:5px 4px 0 8px}.rhlined.hp .meter,.rhfill.hp .meter{stroke:rgba(255,255,255,.4)}.rhlined.hp .medal,.rhfill.hp .medal{fill:#fff}.noBg .rhlined.hp .medal{fill:#919191}.noBg #rh_animcrcl.anim.hp{stroke:#00809d}#rh_animcrcl.anim.hp{stroke:#fff}@media screen and (-ms-high-contrast:black-on-white){.rhlined.hp .medal,.rhfill.hp .medal{fill:#000}#rh_animcrcl.anim.hp{stroke:#000}}#id_rbh.hp{padding-right:12px}#id_rbh.hp.reverse{padding-left:8px;padding-right:0}#id_rbh.hp svg{height:30px;fill:white}#id_rfob.hp{padding-left:12px;color:#fff}.rwds_svg{vertical-align:top;display:inline-block}.rwds_svg.serp{margin:8px 0 0 8px}.rhlined,.rhfill{vertical-align:top;width:32px;height:32px}#id_rh,#id_rbh{position:relative}.id_button toolTip:after,#id_rh:after{white-space:pre-line !important;width:200px}.rhcoinflip{vertical-align:top;width:60px;height:60px;position:absolute;right:-6px;top:-6px}#idCont .rhcoinflip{position:absolute;right:-5px;top:-9px}#rh_meter{vertical-align:top;width:40px;height:40px;margin-left:-36px;margin-top:-4px}.rh_reedm .rhlined,.rhfill,.rh_reedm .meter,.rd_hide{display:none}.rhlined,.rh_reedm .rhfill,#rh_meter{display:inline-block}.noBg .rhlined.hp .meter,.noBg .rhfill.hp .meter,.rhlined.serp .meter,.rhfill.serp .meter{stroke:rgba(177,177,177,.4)}.rhlined.serp .medal{fill:#919191}.noBg .rh_reedm .rhfill.hp .medal,.rh_reedm .rhfill.serp .medal{fill:#00809d}#rh_animcrcl{fill:none;strok [... cut ...] %2011.5948%2017.5332%208.4635%2016.9327%204.77459L16.9212%204.77361C16.7669%202.66391%2015.0063%201.00026%2012.8572%201.00026C11.0684%201.00026%209.54871%202.15296%209.00053%203.75611C8.45243%202.15282%206.93271%201%205.14373%201C2.99462%201%201.23408%202.66369%201.07977%204.77341L1.06734%204.77448C0.466825%208.46338%204.02705%2011.5947%206.00018%2013.0531C7.516%2014.1735%208.59935%, error=curl:Timeout was reached:
Operation timed out after 10006 milliseconds with 25217 bytes received
[Error 28]

DankCloudz, 2023-12-22:
that was multi thread (above)

below is single thread test

Thu Dec 21 20:59:44:start_test_sequence
Thu Dec 21 20:59:44:test bing start...
Thu Dec 21 20:59:44:[test_bing]: url=http://www.bing.com/
Thu Dec 21 20:59:44:test bing ip resolved = 13.107.21.200
Thu Dec 21 20:59:44:test bing ok
Thu Dec 21 20:59:44:test bambulab start...
Thu Dec 21 20:59:44:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.08.02.56&guide_version=00.00.00.01
Thu Dec 21 20:59:44:test bambulab ip resolved = 104.18.3.216
Thu Dec 21 20:59:44:test bambulab ok
Thu Dec 21 20:59:44:test storage start...
Thu Dec 21 20:59:44:[test_oss]: url=http://upload-file.bambulab.com
Thu Dec 21 20:59:46:test storage ok
Thu Dec 21 20:59:46:test storage upgrade start...
Thu Dec 21 20:59:46:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Thu Dec 21 20:59:47:test storage upgrade ok
Thu Dec 21 20:59:47:test storage download start...
Thu Dec 21 20:59:47:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.08.02.00
Thu Dec 21 20:59:50:downloading 10%
Thu Dec 21 20:59:50:downloading 20%
Thu Dec 21 20:59:51:downloading 30%
Thu Dec 21 20:59:51:downloading 40%
Thu Dec 21 20:59:52:downloading 50%
Thu Dec 21 20:59:52:downloading 60%
Thu Dec 21 20:59:53:downloading 70%
Thu Dec 21 20:59:53:downloading 80%
Thu Dec 21 20:59:54:downloading 90%
Thu Dec 21 20:59:55:downloading 100%
Thu Dec 21 20:59:55:test storage download ok
Thu Dec 21 20:59:55:test plugin download start...
Thu Dec 21 20:59:55:[test_plugin_download]: downloadurl=https://public-cdn.bambulab.com/upgrade/studio/plugins/01.08.02.01/363255fecb/win_01.08.02.01.zip
Thu Dec 21 20:59:56:downloading 5%
Thu Dec 21 20:59:57:downloading 10%
Thu Dec 21 20:59:57:downloading 15%
Thu Dec 21 20:59:58:downloading 20%
Thu Dec 21 20:59:58:downloading 25%
Thu Dec 21 20:59:58:downloading 30%
Thu Dec 21 21:00:01:downloading 35%
Thu Dec 21 21:00:05:downloading 40%
Thu Dec 21 21:00:05:downloading 45%
Thu Dec 21 21:00:05:downloading 50%
Thu Dec 21 21:00:06:downloading 55%
Thu Dec 21 21:00:06:downloading 60%
Thu Dec 21 21:00:06:downloading 65%
Thu Dec 21 21:00:07:downloading 70%
Thu Dec 21 21:00:07:downloading 75%
Thu Dec 21 21:00:07:downloading 80%
Thu Dec 21 21:00:08:downloading 85%
Thu Dec 21 21:00:08:downloading 90%
Thu Dec 21 21:00:09:downloading 95%
Thu Dec 21 21:00:09:downloading 100%
Thu Dec 21 21:00:09:test plugin download ok
Thu Dec 21 21:00:09:end_test_sequence

DankCloudz, 2023-12-22:
I just tried TWO earlier versions of Bambu Studio and it still will not connect to my A1 Mini. (ver. 1.07.07.89   &  ver. 1.08.00.62)
/sigh Day 4 without being able to use my A1 Mini :(

[debug_Fri_Dec_22_16_37_20_12752.log.zip](https://github.com/bambulab/BambuStudio/files/13756872/debug_Fri_Dec_22_16_37_20_12752.log.zip)

Please fix this. I am counting on you so when Big Daddy X2 comes out I can confidently get one. I may still get the full size A1 if this situation gets situated timely.

DankCloudz, 2023-12-22:
I would like to downgrade the FW on my printer. Please assist. 
It is currently reading FW versions: (OTA:01.01.02.00) (AMS:00.00.07.89) from the printer settings.
It will not connect to my Bambu account in the printer.

DankCloudz, 2023-12-23:
Apparently, after hours of messing with my printer, home internet network, the Bambu Handy app on my phone, and the Bambu Studio PC app, I was finally able to resolve my issue on my own. 

Some insight...
This problem may have arrised when I renamed my printer in the app from the stock name it came with to "A1 Mini".
I realized the issue is not on the Slicer side because the printer would not load any account and refused to populate a QR code to "link with account".
I noticed when attempting to Link to account, that there was my renamed printer giving me the option to attempt to link, and the originally named printer (already linked but unresponsive). Yes two printers, 1 named "A1 Mini" (which i recently named) and the Originally named printer (same one) "3DP-123-456" (not actual name).
I then unlinked the previously correct one ("3DP-xxx-xxx") from my Bambu Account.
I also moved the printer to a few feet away from my network router.
I renamed the printer back to the original name (3dp... etc) to try to avoid confusion.
At this point, after a few attempts, I was able to get the printer to link to my Bambu account.
Confirmed with the Bambu Handy app that the printer was communicating and the device tab was reading bed/nozzle temps.

This took several factory resets, forgetting wifi settings, rerunning the calibrations every time a factory reset was done. 
I will provide as much info as possible to help other users avoid this frustrating issue. 
I am also a tad disappointed with encountering this issue and having to resolve it myself, without hearing back from BBL for over two days, when providing the requested information.
Guess I'm not buying a full size A1.

walterwongbbl, 2023-12-25:
Hi. @DankCloudz 
I apologize for the inconvenience. We happened to be on holiday, and that's why we missed your message. I'm glad to hear that you have resolved the issue. After analyzing your logs, it seems that the main cause of the problem is that the studio was attempting to establish a connection with our server but failed. This issue is mostly caused by network environment factors. You can verify this problem by using your mobile hotspot. Additionally, I will escalate this issue to the printer department and request their analysis.

closed 2024-01-15 by walterwongbbl

AlekseiKV, 2024-04-01:
Hi! I found the cause of the problem and how to solve it. You just need to format the flash drive))))))