bambulab/BambuStudio #3189: A1 extra toolhead travel move on layer 3 with timelaps disabled

Naninani, 2023-12-21:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS Sonoma 14.2

### Printer

Bambu Lab A1

### How to reproduce

Enable the travel moves in the slicer to observe the behavior. 
A1 does an extra move to the left side of the bed when the timelaps is disabled.
It always happens exactly on layer 3.

With timelaps gcode in the extruder:
[image]

When the timelpas gcode is removed from the extruder settings:
[image]

On top of that the extra movement adds print time.

### Actual results

A1 does an extra move to the left side of the bed when the timelaps is disabled/not used.
It always happens exactly on layer 3.
I observed this on each print I do.

### Expected results

The printhead should not do any extra movement related to timelaps on this specific spot when timelaps is not used. I'm not even sure if its supposed to go there.
If timelaps is enabled, the printhead should move only to the poop shoot.

### Project file & Debug log uploads

No need to attach file, simple enable the travel moves in the slicer to observe the behavior. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

terbospaghetti, 2024-01-13:
I am attempting to implement this gcode fix, but I am unable to find the gcode dialog box in your pictures in order to modify it. I have tried turning on "advanced" as well as "developer" modes, but I still have found no sign of the "printer settings" page you display here.

Where should I be looking??

SaltWei, 2024-01-17:
This is used to check whether the nozzle is warped by filament. Not bug

Naninani, 2024-01-18:
@SaltWei, If that is the case why it is in the timelapse code? Imo this move is not necessary and could cause layer shifts and perimeters blobs.

SaltWei, 2024-01-19:
> If that is the case why it is in the timelapse code?

For A1 and A1 mini, slicer will try to put "timelapse gcode" before printing infill, so the filement that spills out of the nozzle while travel can be absorbed by infill.

And the gcode to move away from model to check whether the nozzle is warped by filament is same. So it has same position with timelapse gcode.

The developer did one lazy thing to reuse the gcode block.

Naninani, 2024-01-19:
@SaltWei that may work when there is infill. However, if the model is only of perimeters wall or in vase mode this is becoming a problem.
Also check the discord community and how many people are complaining about this travel move.

I personally removed the code as it was causing layer shifts.

mreynolds0404, 2024-01-25:
What does "to check whether the nozzle is warped by filament" mean?

SaltWei, 2024-01-25:
> What does "to check whether the nozzle is warped by filament" mean?

During printing, the printer move to side and try to touch the bed to check whether the nozzle is covered by filament which may cause nozzle clog or damage

mreynolds0404, 2024-01-25:
These moves appear to be ruining print quality. Is it safe to disable them?
[image]
[image]

mreynolds0404, 2024-01-25:
I printed a string test model and the slicer showed the same travel moves but none of them were taken during the print, and the print came out fine. Unsure how to interpret.

[
[image]
](url)

goyetus, 2024-02-09:
I have opened a new BUG about this problem....

After a POWER LOSS ,  The printer STARTS to do all the PARK MOVEMENTS.
The printer ignored the "no timelapse"  config,   and  continue all down doing ALL movements.....  really bad.

https://github.com/bambulab/BambuStudio/issues/3567

davidpmclaughlin, 2024-04-14:
I found the X-4.2 that is causing my shifting of about 4.2 mm. I replaced it with X0 and it seems to work, but the whole thing is probably not necessary.