bambulab/BambuStudio #3190: Error Code 5010 when sending jobs

sawig10, 2023-12-21:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1c

### How to reproduce

Send a plate greater then 3.3MBs and at 99% it will fail. 
Have to delete the gcode that appears corrupted now and try again. 

### Actual results

[image]

### Expected results

The sliced job should be sitting on the machine read to print or be selected to print from SD card.

### Project file & Debug log uploads

[log122123_error5010.zip](https://github.com/bambulab/BambuStudio/files/13736780/log122123_error5010.zip)
[AMS Tall Stand V2.zip](https://github.com/bambulab/BambuStudio/files/13736785/AMS.Tall.Stand.V2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

sawig10, 2023-12-21:
I have logged out of both the studio and off the printer.
Reformatted the SD Card.
Factory reset the printer.
Uninstalled and reinstalled Studio. 
Turned firewalls off.
Added an assess point in the network thinking it was low signal. 
Resync server time and changed the time server. 
Note: Handy App has no issues with connection.

walterwongbbl, 2023-12-21:
Hi. @sawig10 
I successfully attempted to send the file to the printer using the file you provided. Additionally, I reviewed the log files you provided and noticed several timeout records related to FTP operations. Please try placing the printer closer to the Wi-Fi signal and then attempt the send again. FTP has higher requirements for network signal stability.

sawig10, 2023-12-22:
@walterwongbbl your FTP comment brought to mind that I was previously testing other FTP issues not related to Bambu SW. I turned off/blocked some of those protocols and returning those network settings fixed my problem. I am no longer getting kicked on and off using the Studio.

walterwongbbl, 2023-12-22:
@sawig10 
Is your issue resolved now?

sawig10, 2023-12-22:
Yes, looks like issue has been resolved. Thank you!

closed 2023-12-25 by walterwongbbl

sawig10, 2024-06-06:
> Hi @sawig10<https://github.com/sawig10> , I confront same issue as you (Error code -5010 , stuck at 99%).
> Could you share what fix you did to fix this issue ? Many thanks.

Hey @CCKang886, I found that enabling FTP, TLS, and Samba on my router seemed to resolve my problems. I also changed my DNS settings around the same time so they may have played a factor as well.

CCKang886, 2024-06-06:
Hi @sawig10 , I will try fix following your guidance. Will feedback with result. Thanks !