bambulab/BambuStudio #3194: Auto arrange feature cant arrange large objects

ochiburi, 2023-12-21:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

MacOs Sonoma 14.1.2

### Printer

Bambu Labs P1S

### How to reproduce

1. Create any object whose dimensions are close to the size of the build plate - eg. 240x240 - in the attached project file see plate 03, Object named Bracket
2. Put the file on a build plate
3. See the warning "objectname is too close to exclusion area, there may be collisions when printing."
4. Press the plate autoarrange button
5. Object is moved off plate.
6. Put back manually.
7. Make minor corrections via mouse/keyboard until warning disappears. (Thus object is now outside exclusion areas).
8. Just for fun press plate autoarrange button again.
9. Object is moved off plate again!

### Actual results

Object is moved off plate

### Expected results

Object should be auto arranged and placed on plate without crossing exclusion zones

Please also provide visible guides on the plate showing exclusion zones to aid manual positioning

### Project file & Debug log uploads

[Lantern.3mf.zip](https://github.com/bambulab/BambuStudio/files/13741944/Lantern.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

RaphTronic, 2024-01-04:
Same xperience here. I print a lot of rectangular flat models that are only a few mimllimeters shorter than the plate width. Auto arrange regularly fails to place them, although a manual placement works. Which is tedious because it is very dificult to see the margin left between a part and the ege ofthe plate because of the big white border surrending the selected part.
There is no reason why auto arrange should fail in such very simple cases.

tangketan, 2024-01-17:
Thanks for reporting the issue. We'll fix it in the next big version (1.9).

closed 2026-11-03 by ArthurBambulab