bambulab/BambuStudio #3196: Rotation center should be mounce pointer instead of plate center

caco3, 2023-12-21:
FreeCAD (and most likely various other tools) has a nice rotation feature: The center of the rotation is where the mouse points to. Bambu Studio in opposite always rotates around the center of the plate. This is very inconvenient if an object is not centered or far zoomed in.

Related issue: https://github.com/bambulab/BambuStudio/issues/3044

RaphTronic, 2024-01-04:
Incredibly irritating behavior, indeed.
Especially when slicing tall parts as it becomes impossible to orbit around details at height without dozens of zoom-outs / orbit / zoom-ins.
All the CAD programs I use orbit around the center of the screen or position of the mouse.
Bambu Studio is the only application I use with this behavior.

Like the "Zoom to mouse position" preference setting in Studio, there should be an "Orbit around mouse position" or "Orbit around center of screen" setting.

RheaAyase, 2024-12-18:
This, so much this, plus the rotation in the painting mode is just braindead unusable. Imagine selling 4xAMS for 16 colour printing but your software doesn't support reasonable painting...

Argon5504, 2026-11-15:
Coming from Blender, the rotation to mouse is very helpful in navigating, its sad to see this issue open to this date and hasn't been implemented.