bambulab/BambuStudio #3198: Strange Wipe Tower Off Build Plate

nworbcire, 2023-12-22:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11 Pro

### Additional system information

OS Name	Microsoft Windows 11 Pro
Version	10.0.22621 Build 22621
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	DESKTOP-8LT6N1D
System Manufacturer	Alienware
System Model	Alienware m15 R3
System Type	x64-based PC
System SKU	09CE
Processor	Intel(R) Core(TM) i7-10875H CPU @ 2.30GHz, 2304 Mhz, 8 Core(s), 16 Logical Processor(s)
BIOS Version/Date	Alienware 1.24.0, 7/6/2023
SMBIOS Version	3.2
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	Alienware
BaseBoard Product	Alienware m15 R3
BaseBoard Version	A00
Platform Role	Mobile
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume1
Locale	United States
Hardware Abstraction Layer	Version = "10.0.22621.2506"
User Name	DESKTOP-8LT6N1D\ericc
Time Zone	Eastern Standard Time
Installed Physical Memory (RAM)	32.0 GB
Total Physical Memory	31.8 GB
Available Physical Memory	16.3 GB
Total Virtual Memory	44.8 GB
Available Virtual Memory	24.2 GB
Page File Space	13.0 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	On
Virtualization-based security	Not enabled
Windows Defender Application Control policy	Enforced
Windows Defender Application Control user mode policy	Off
Device Encryption Support	Elevation Required to View
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes

### Printer

X1

### How to reproduce

Using variable layer height options. Sliced several models "painting" my own heights then clicked "adaptive" using "0.02" and then sliced model resulting in this...

[image]

### Actual results

Slicer finishes, wipe towers heads off the plate???

### Expected results

Wipe tower should NOT go off the build plate

### Project file & Debug log uploads

[Debug.zip](https://github.com/bambulab/BambuStudio/files/13747814/Debug.zip)

Project file too large to send (31MB)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-12-27:
@nworbcire  Could you compress the 3mf file into zip format and then upload it?

zhimin-zeng-bambulab, 2024-01-17:
@nworbcire  Github can upload files no larger than 100MB. You can compress it into a zip and then upload it.

zhimin-zeng-bambulab, 2024-02-04:
@nworbcire  Could you share the 3mf file to help us solve the problem?

closed 2024-08-20 by zhimin-zeng-bambulab