bambulab/BambuStudio #3199: Video Loading Failed [-90]

Axsgranted, 2023-12-22:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Don't know what caused the issue, only that it has occurred.  Restarting Bambu Studio, in the middle of the print, still did not allow me to load the video.

### Actual results

[image]

Received Video Loading Error [-90]

### Expected results

Load the video

### Project file & Debug log uploads

[Project Logs.zip](https://github.com/bambulab/BambuStudio/files/13747891/Project.Logs.zip)
[OnePieceKeyChainProject.zip](https://github.com/bambulab/BambuStudio/files/13747893/OnePieceKeyChainProject.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Axsgranted, 2023-12-22:
Confirmed the printer is online.  I can communicate with it, see current printing status, etc, nothing but video fails.

Axsgranted, 2023-12-22:
[image]

Superdad1106, 2023-12-23:
I too am having this issue. I'm not sure how to get any bug info or anything like that but this is the exact same issue I have. Video wont load in Studio or in Handy. Just get loading failed [-90]. Seems to have happened after updates to both printer and studio if I remember correctly. Printer still online. Can send prints to printer, watch the progress, time remaining, percentage all still work and show. Just not the video. I'm assuming since the video doesn't work the spaghetti detection and other features wont work correctly. Will check to see if timelapses still record to SD after this print is done. I'm assuming camera still works but something is going on with the transmission. Any updates or useful information would be greatly appreciated.

Axsgranted, 2023-12-23:
I was able to resolve the issue by simply turning the printer off and on again, but I still don't know what conditions put it into that state.

Superdad1106, 2023-12-23:
> I was able to resolve the issue by simply turning the printer off and on again, but I still don't know what conditions put it into that state.

WOW thanks for the fast response! I’m pretty sure I cycled the power to my printer once or twice since I noticed the issue but I could be wrong, it’s possible even if I did, Studio was still open on my PC though. When this print is done I will try that again and make sure everything else is closed and see if it works. Hopefully it’s that easy to fix.

Superdad1106, 2023-12-24:
Just got around to being able to restart everything and that worked! Thank you.

seva-luchianov, 2024-01-15:
This is happening to me as well, both in Handy and in studio. Not sure what triggers it, currently I'm experiencing it mid-print (video feed was working in the beginning)
Obviously I cannot restart the printer but generally restarting it does seem to fix it at least for a day or two

rache6987, 2024-01-15:
Same issue. Happened twice today on X1C. First time it went down after approx 5 hrs of working normally. After the first time, I restarted my printer because print had spaghetti failure (was not alerted) and it was working normal again. Started print again and same -90 error approx 1 hour in.

Axsgranted, 2024-01-16:
Adding some additional behaviors I've seen:  After the video fails to load with this error I've noticed:

1. I'm unable to retrieve data from the MicroSD Card via the Slicer.  Connection timeout error
2. Printing progress does update but then nothing else does.  Controls, Spool data, etc.
3. Update page is also in accessible

This led me to believe that the perhaps there was a wifi connection problem, but I checked my access point logs and it never lost connection (uptime for over 3 days).  And the printing status, does still update.

To me, this points to some service on the device itself that crashes and then times out, which is why it requires a hard restart of the device before it functions again.

Axsgranted, 2024-01-17:
Update: Downloaded the latest version of Bambu Studio, 1.8.4.51 and I was presented with the same -90 error again.  However one big difference, I let it continue to attempt to reload as I've done with previous versions, and within a few minutes it was successful and the video feed was back up!

DruidSGribami, 2024-01-18:
Hi! Same problem with camera, loading failed -90, it happens after firmware update 01.07.02.00, before an update it works perfectly fine. Printer restart - camera start working, when I send another job to printer - same issue with camera.

DruidSGribami, 2024-01-19:
> Hi! Same problem with camera, loading failed -90, it happens after firmware update 01.07.02.00, before an update it works perfectly fine. Printer restart - camera start working, when I send another job to printer - same issue with camera. After my print will be finished I try to roll back firmware to previos version with Handy app.

Update 19.01.24 - roll back firmware to 01.07.01.00 ver., right now camera works fine, I have video on Handy app and in Bambu studio via internet, so its new firmware problem I think.

bambu123, 2024-01-19:
@DruidSGribami @Axsgranted @rache6987 
We need the printer log.
Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).
After that, give me the ticket No. here.

DruidSGribami, 2024-01-19:
> @DruidSGribami @Axsgranted @rache6987 We need the printer log. Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log). No need update the printer firmware and reproduce because old log in several days is kept in sdcard. After that, give me the ticket No. here.

Create a Ticket No: US240119506001

RyanWor, 2024-01-22:
> Update 19.01.24 - roll back firmware to 01.07.01.00 ver., right now camera works fine, I have video on Handy app and in Bambu studio via internet, so its new firmware problem I think.

My X1C is still on 01.07.01.00 FW (have not installed latest update) and I am also having this same issue. Camera works fine for a while, then stops loading after a while and won't work again until full power off and back on of the printer.

bambu123, 2024-01-23:
@RyanWor @seva-luchianov @Axsgranted @Superdad1106 
For all of you, 
In addition to restarting the printer, please also try restarting Studio to see if there are any changes.

seva-luchianov, 2024-01-23:
I've been power cycling the printer daily and that's been enough to keep the issue at bay. Haven't attempted to repro it since piping up about it here.

bambu123, 2024-01-23:
@RyanWor @DruidSGribami 
How about the liveview with mobile phones?

RyanWor, 2024-01-23:
> @RyanWor @seva-luchianov @Axsgranted @Superdad1106 
> For all of you, 
> In addition to restarting the printer, please also try restarting Studio to see if there are any changes.

From what I've tried restarting/force close for Studio/Handy has no effect on the issue. When the camera stops working it's no longer visible in any of the apps across all my devices. I have to fully power off the printer for a few moments and then power it back on to get the camera working again.

bambu123, 2024-01-24:
> > @DruidSGribami @Axsgranted @rache6987 We need the printer log. Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log). No need update the printer firmware and reproduce because old log in several days is kept in sdcard. After that, give me the ticket No. here.
> 
> I created a Ticket No: US240119506001 log file inside 00M09A3B1400620_all_20240119143038_enc.tar.gz

Sorry, I forgot to mention that we also need Studio logs. Please help attach the Studio logs for the time range of the problem to the ticket. Let's analyze it against the printer logs as soon as possible.

bambu123, 2024-01-24:
> From what I've tried restarting/force close for Studio/Handy has no effect on the issue. When the camera stops working it's no longer visible in any of the apps across all my devices. I have to [...]

Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).
Please also help attach the Studio logs for the time range of the problem to the ticket.

Axsgranted, 2024-01-24:
> @RyanWor @seva-luchianov @Axsgranted @Superdad1106 For all of you, In addition to restarting the printer, please also try restarting Studio to see if there are any changes.

Per the notes when I opened this issue, restarting Studio does nothing.  Only turning the printer on and off again seemed to work.

Axsgranted, 2024-01-24:
> @RyanWor @DruidSGribami How about the liveview with mobile phones?
> 
> Please help provide connection-related logs. These logs are not output by default. You need to create a "tutk_log" subdirectory under the log directory to output them. These logs will include your device IP. It is recommended to send them to me via email.
> 
> %APPDATA%\BambuStudio\log\tutk_log
> 
> When you create the "tutk_log" directory, you need to restart Studio and reproduce the problem. It is best to wait for a longer playback time, more than 1 minute.

Sent to your email this morning (PST time).

DruidSGribami, 2024-01-24:
> Sorry, I forgot to mention that we also need Studio logs. Please help attach the Studio logs for the time range of the problem to the ticket. Let's analyze it against the printer logs as soon as [...]

log.7z attached to Ticket No: US240119506001 Ticket No: US240119506001 here bambu studio logs all i have in folder, yesterday i had problem again (23 Jan)

DruidSGribami, 2024-01-24:
> @RyanWor @DruidSGribami How about the liveview with mobile phones?
>
> Please help provide connection-related logs. These logs are not output by default. You need to create a "tutk_log" subdirectory [...]

I sent yesterday 23 Jan via mail chunmao.guo@bambulab.com

Axsgranted, 2024-01-24:
> @DruidSGribami @Axsgranted @rache6987 We need the printer log. Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log). No need update the [...]

Opened a ticket and uploaded relevant printer and studio log files.  Ticket# US240124612001

bambu123, 2024-01-25:
> Opened a ticket and uploaded relevant printer and studio log files. Ticket# US240124612001

From the logs, we found that this is a known camera service crash issue on the printer side. Users need to wait for version 1.8 to fix it. Currently, they can only restart the printer when there is a problem.

Axsgranted, 2024-01-25:
> From the logs, we found that this is a known camera service crash issue on the printer side. Users need to wait for version 1.8 to fix it. Currently, they can only restart the printer when there is [...]

Called it! :)

hackitz, 2024-01-25:
just want to add my experience, my camera gives me the load failed -90 after 3 or 4 prints. if i remove the sd card and plug it into windows i get a prompt the drive has errors. i let windows repair the sd card and when put back into the x1c it works as expected.
something is corrupting my sd card.

seva-luchianov, 2024-01-25:
> From the logs, we found that this is a known camera service crash issue on the printer side. Users need to wait for version 1.8 to fix it. Currently, they can only restart the printer when there is [...]

Thank you for digging into this issue and finding the root cause! I was getting nowhere through the official support ticketing gauntlet. Looking forward to v1.8 :D

Also +1 to what @hackitz said above, I had an identical experience when first troubleshooting this problem.

bambu123, 2024-01-26:
> @bambu123 Actually I'm on version 1.07.02, when you say 1.8 do you mean 1.08? :)

Yes. 01.08 .The next major version.

SsjDennis, 2024-02-07:
I know I'm a bit late to the conversation but I have exactly the same problem as mentioned above. Printer is connected but video feed and then SD card acces doesn't work. Powercycling the Printer does fix the issue for a while. 
I do want to add, that I never had this problem before, when I didn,t have an sd Card in the printer. It may be related to the problem.

grtessman, 2024-02-07:
Got this issue tonight "Loading failed [-90]!" for the first time after getting Network Plugin update.   Sigh.

Network tests passed, and mobile says "Timeout after 00:00:05.000000: start tutk server timeout"  I'm guessing it occurred because I started a print job and as it was sending to the cloud, I clicked Cancel in the windows 1.8.4.51 and the app hung and I closed, relaunched and resent my print job with small corrections.  Likely a reboot will help but it's because something on the cloud side isn't letting go of the previous print job maybe?

Then it started acting up with this error.  Likely a restart of the printer may help fix it but it's too late now - print job already started.  Dang!

MonsterMaxx, 2024-02-17:
Mine has the same "Load failed [-90]" issue and power cycling it did NOT correct it.

bambu123, 2024-02-18:
> Mine has the same "Load failed [-90]" issue and power cycling it did NOT correct it.

Is it an X1 printer? Is it an occasional -90 error, or is it always like this?

MonsterMaxx, 2024-02-18:
> > Mine has the same "Load failed [-90]" issue and power cycling it did NOT correct it.
> 
> Is it an X1 printer? Is it an occasional -90 error, or is it always like this?

Started a few firmware upgrades ago, been going on several weeks now.  It's always like this now :(

X1C printer.

bambu123, 2024-02-18:
> > > Mine has the same "Load failed [-90]" issue and power cycling it did NOT correct it.
> > 
> > 
> > Is it an X1 printer? Is it an occasional -90 error, or is it always like this?
> 
> Started a few firmware upgrades ago, been going on several weeks now. It's always like this now :(
> 
> X1C printer.

Can it be viewed on mobile phone with Handy?
Have you check the network connection?
Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).
Please also help attach the Studio logs for the time range of the problem to the ticket.

MonsterMaxx, 2024-02-18:
> Can it be viewed on mobile phone with Handy? Have you check the network connection? Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log). [...]

Nope, Handy is out too.  The network connection is fine, that's how I'm sending prints to the printer.

I'll work on submitting a ticket tomorrow

JaminJimmy, 2024-02-22:
I am having the very same error with my X1C and it seems to have started after getting some network module update.  I have to admit I don't shut down or restart the machine everyday like I probably should but bambu broke this they need to fix it.  I have put in a ticked for this issue as well and hopefully they will unbreak it with another pushed network update.

bambu123, 2024-02-23:
> I am having the very same error with my X1C and it seems to have started after getting some network module update. I have to admit I don't shut down or restart the machine everyday like I probably [...]

Networking plugin update can't fix this. Please wait for X1C firmware update.

Wixely, 2024-03-22:
I'm also getting this issue. FW 01.07.02.00.
Restarting the machine only works 50/50 and when it does, I usually get one stream before it breaks again.
Is there any timeline on a new FW to fix this?

Wixely, 2024-04-08:
I found that for me it's caused by Lan Mode Liveview being enabled on the printer. If I disable this I can stream video. If I enable this, it works for a short period of time, but eventually stops working similar to a machine restart.

MonsterMaxx, 2024-04-08:
Have you tried V1.9 Beta?  Fixes a lot of issues.

AceGrinder, 2024-04-30:
Sign me up as one experiencing this as well. Sometimes multiple times a day.

Poewa, 2024-06-09:
Same issue. Posting for clarity

JaminJimmy, 2024-06-11:
> *Please upload log through Handy APP, and give us your ticket number*
> ------------------------------
>
> *Upload log files through Bambu Handy*
> Log files from the Bambu printer can be uploaded via Bambu Handy over the
> network, with an upload speed of around 150 kbps under optimal network
> conditions.
> When uploading logs, you have the option to select a specific time frame,
> allowing you to reduce the volume of logs transmitted and shorten the
> upload duration.
> To upload files, go to *Me* → *Support Tickets* → Choose the ticket for
> log upload and proceed with the operation.
> [image: app_support_ticket]
> <https://camo.githubusercontent.com/8a9b0094adc6f12470aec9b0e511b068b9287e5db19ba1a6346329b8879d825f/68747470733a2f2f77696b692e62616d62756c61622e636f6d2f70312f74726f75626c6573686f6f74696e672f75706c6f61642d6c6f672f6170705f737570706f72745f7469636b65742e706e67>

This issue has been resolved, I simply had to restart my X1C and the
network driver has been updated and have never seen this issue since.

Thank you,
Jimmy Fallin

sonjamichelle, 2024-07-27:
Firmware version 1.08 was uploaded to my X1C. I've been battling this issue for a few months now.

Here is what I did. I shut off the printer. Unplugged it from the wall for about 30 seconds.
Turned it back on, waited for it to connect.
Then I used the BambuHandy app to update the firmware.
A couple of caveats. Make sure if you are enrolled in the Beta Firmware program to unenroll. This will prevent you from getting 1.08.
You should be able to see 1.08 as an upgrade. do the upgrade. Once it has ugraded at sat for about five minutes, do anoter power cycle.
Would be good to run a device self test and calibration at this point.
So far the camera and the connection to the printer from BambuStudio has been stable for a couple of hours now. Usually I only get about 30 minutes before I lose connection to the camera/printer.

sonjamichelle, 2024-07-30:
OK, looks like I spoke too soon. 1.08 does NOT fix the issue. Camera stays on for a bit longer, but now I get an error that tells me the printer has been logged out, please check the network.

That wording may not be exact, I'm going from my poor memory. Looks like there might be a different issure at play, or at least what ever changes that were made in 1.08 might be leading towards the real issue with better error messages. Just my take from a non-coder's perspective.

I just fired up BambuStudio to double check my wording, here is a capture of the message I am getting now. My printer has NOT been restarted since the initial reboot after the firmware upgrade. I've been printing some brackets needed to finish my workshop and time is short.

[image]

Hopefully we can get a resolution for this. In the mean time I would set up an alternate camera using a Pi or another PC if you need to keep an eye on your print while away. One thing that can be done, OctoEveryWhere has the ability to use an alternate camera that will enable its AI in addition to the main Bambu camera. So there are some optional work arounds, inconvienent yes, but not all is lost.

As long as the main features of the Printer and BambuStudio continue to function printing can still be accomplished. ;-)

sonjamichelle, 2024-08-04:
This is the latest message I get.

[image]

The printer FINALLY sucessfully upgraded to 1.08. Didn't upgrade from the app, but upgraded from the LCD.

The cmera was working great for about 3 days of pritning. I was able to monitor from the other room and remotely from my phone with Bambu Handy. Then as soon as I fired up BambuStudio on my PC after about the second print, the camera issue came back. Coincidence????

dmorris68, 2024-08-11:
So the problem is definitely not fixed in 1.08. I'm getting "The printer has been logged out and cannot connect" error, both in Bambu Studio and Orca Slicer, as well as the Handy app, every day and sometimes multiple times per day. It happens anywhere from about 30 minutes of printing to a few hours, but it consistently happens. I have to power cycle the printer to get the camera back.

PRONOVATEK, 2024-09-30:
I have the same problem, my firmware is 1.08.02 and constantly reports problems with wifi although have connection with bambu handy, I don’t know what more to do…

Scottmg1, 2024-10-26:
Having same issue here on all 4 X1C’s I have. On Bambu Handy or Studio.

Noted that issue didn’t start ever occurring several hours later until LAN Mode Liveview was enabled on the printer touchscreen. 

Any update on this?

Scottmg1, 2024-11-07:
@bambu123 Just bumping this here.

bambu123, 2024-11-07:
It is recommended to restart the printer and note that the printer can support limited liveview connections at the same time.

Scottmg1, 2024-11-07:
@bambu123 I would like to note their is only ONE liveview connection over LAN to the X1C. And this issue only occurs when the LAN Mode Liveview is enabled on the touchscreen of the X1C. Every 4-5 hours the camera stream will stop working until the printer is restarted like you said. 

So my question is that the function of the X1C Camera with only 1 Liveview connection breaks every 4 hours of the printer being on and that I should turn off and on my X1C during a print so I can have a functioning camera? This is not normal operation. This issue is present on every single one of my X1 Carbons.  

Please confirm with the team if this is a bug? This is more a bug with the printer firmware than it is with Bambu Studio.

bambu123, 2024-11-07:
What error is displayed on the User Interface? You can provide a screenshot of the error and Studio logs for us to analyze.

sonjamichelle, 2024-11-07:
For me lately, the camera has been "behaving". Pretty much the ONLY thing "behaving" on the printer at the moment. I've been having a lot of issues with the tool-head grabbing "poop" out the chute and dropping them all over the bed causing the AI to pause the print because of a perceived error. Though the blasted "AI" will complete miss a REAL failure like it did last night while the printer was going when I was sleeping. Woke up to a right mess instead of a really finished print.

No poop boogers on the plate, but the print adhesion failed and just left a stringy mess on the bed. Go figure, AI isn't as "intelligent" as everyone makes it out to be, lol

As for the camera, I think that has to be tied to the reported TCP/IP stack instability many folks have reported in the community.

CCS86, 2024-12-04:
@bambu123 

[image]

bambu123, 2024-12-05:
Restart the printer and try again。

CCS86, 2024-12-05:
> Restart the printer and try again。

The camera was actually working fine. Then I had to power cycle the printer and after that the camera will not connect. The printer has full connectivity otherwise and passes the network test.